AlexandreRouma/SDRPlusPlus #974: Option to disable AGC entirely

TS440S, 2023-01-24:
Could an option be added to completely disable the demodulator AGC? GQRX has this feature and it has been very useful to me when trying to record specific signals with a fluctuating strength. I tried setting the attack and release levels but if still always eventually adjusts the gain.

If this is already an option, where is the setting?

AlexandreRouma, 2023-01-25:
Why would you want no AGC on a signal with a changing level?
I feel like you're using the demodulators for something they're not meant to.

Georg1775, 2023-01-28:
I'd like to have this option, too. If I listen to a radio sonde signal (USB) or to slow CW signals (i.e. signals with short "data" and a longer pause in between) I always hear the noise level increasing, at least shortly before the signal starts again. This happens even with attack on maximum and decay on minimum value. For CW, at least for me, it makes it harder to recognize the beep lengths because the beginning of every beep is a little bit smoothed by the down regulating AGC.

TS440S, 2023-01-28:
Kiwi SDR has the option to disable it entirely and I really prefer that. As a signal changes in strength, the software adjusts the volume which makes the noise sort of pop up between strong and weak portions of the signal. Like the person above said, CW will do this, an AM radio station pause will do this too. I'd just rather have the option to disable it entirely.

kord1e, 2023-02-12:
I support this idea! No AGC and manual level control is sometimes useful.

delboy85341, 2023-02-28:
> Why would you want no AGC on a signal with a changing level? I feel like you're using the demodulators for something they're not meant to.

I have a loop antenna for MW and adjusting the variable capacitor for optimum results on any particular frequency would be easier with agc  disabled.  I tune to a station and adjust the capacitor for the loudest sound.  No agc would make it easy for the loudest sound, thus the strongest signal.

AlexandreRouma, 2023-03-01:
Just look at the peak on the FFT, it'll be way more accurate than juding how loud the output is...

closed 2023-03-01 by AlexandreRouma