AlexandreRouma/SDRPlusPlus #977: SDR++ do not show "SDR++ server" in source.

git-minix, 2023-01-29:
**Hardware**
- CPU: AMD Ryzen 7 2700X
- RAM: 32 GB
- GPU: Radeon R9 290/39
- SDR: AirSpy HF+ Discovery remote via sdrpp --server on Raspberry Pi 3

**Software**
- Operating System: Debian 11.6 
- SDR++: Version + Build date 1.1.0-913

**Bug Description**
Following https://www.youtube.com/watch?v=Umr8OA62AKA there should be a menu item "SDR++ server" under "Source". It is not. Not sure if there is a hidden config option needed to activate this feature. By the way the connect to a spyserver is working fine via "Source" menu.

**Steps To Reproduce**
1. start SDRpp via command line 
2. check the SDRpp app for connect option under "Source"

git-minix, 2023-01-29:
Also other users are impacted with the same issue - see comments in 

https://www.youtube.com/watch?v=Umr8OA62AKA

"Hello. I am running a server on mac OS and Client on Windows 10. I have the January 26th nightly release. I cannot find SDR++ server in sources. Is there any other option I should try?"

AlexandreRouma, 2023-01-29:
Time to RTFM: https://github.com/AlexandreRouma/SDRPlusPlus#i-dont-see--insert-module-name-here--whats-going-on

closed 2023-01-29 by AlexandreRouma

git-minix, 2023-01-29:
thanks - deleting /.config/sdrpp/config.json and restarting sdrpp was the solution

logging-synchronous, 2023-08-13:
RTFM: https://github.com/AlexandreRouma/SDRPlusPlus#i-dont-see--insert-module-name-here--whats-going-on

deleted config.json, restarted sdrpp, all is well.