AlexandreRouma/SDRPlusPlus #979: Time shift rec/play

FredHucht, 2023-01-29:
Hi and thanks for this excellent software!

I have a feature request that I already posted in the discussions section 2 weeks ago, without much resonance. 
It might be quite easy to implement:

If SDR++ would save the whole baseband data into a ring buffer of the size of, say, the waterfall, one could implement "Time Shift rec/play": By clicking into the waterfall at a certain time in the past, the decoding and playback would happen at that time instead at the actual time. The playback time could be marked with a horizontal line. With such a feature, it would be easy to listen to short radio comms as in air radio. Furthermore, one could "see the future" and jump to signals in advance.

The memory requirements should be ~500MB/min at 2.048 MHz bandwidth, so several minutes should fit into typical memory.

Best, Fred

AlexandreRouma, 2023-01-29:
It was tested in the early days of SDR++ and was too much of a pain to implement cleanly in addition to not being that useful. For high rate SDRs (aka anything above a RTL-SDR), the amount of memory needed quickly goes up.

This will not be implemented.

closed 2023-01-29 by AlexandreRouma

illtellyoulater, 2024-01-29:
Back then I really enjoyed using SDR# Timeshift plugin, as it made my initial listening sessions a bit less goofy xD and action packed! I think it's super cool when you are just starting and don't really know much about the RF space ;)

And even when I got a bit more experienced, I still found this approach funny and useful, as it helped me make the most of those random and unpredictable short moments we sometime only have to dedicate to this intriguing technology area. 

So @AlexandreRouma I really hope you can reconsider your decision... Of course this should be implemented in a very optimized and memory-efficient way, for example making this timeshift only available on narrower bandwidth sampling settings  - or above a certain zoom level - or it could be dynamically parameterized based on system resources.

In other words, I'd really love to see something like this come alive in a new, efficient and lean approach, in line with the code quality of this project... It could actually be a stimulating programming task itself...