AlexandreRouma/SDRPlusPlus #986: Nightly does not work on Ubuntu 22.04

drewlio, 2023-02-08:
Looks like it needs libvolk.so.2.2 but libvolk.so.2.5 is current. This is on today's nightly.

```
$ sdrpp
sdrpp: error while loading shared libraries: libvolk.so.2.2: cannot open shared object file: No such file or directory

```

```
/usr/lib/x86_64-linux-gnu/libvolk.so
/usr/lib/x86_64-linux-gnu/libvolk.so.2.5.1
/usr/lib/x86_64-linux-gnu/libvolk.a
/usr/lib/x86_64-linux-gnu/libvolk.so.2.5
```

drewlio, 2023-02-08:
PR #919 knew this was going to happen.

AlexandreRouma, 2023-02-08:
That pull request has nothing to do with what you're experiencing, you probably just downloaded the wrong deb for your Ubuntu version. You need to use the `jammy` deb.

drewlio, 2023-02-08:
You might be right. I'll check. Thanks.

closed 2023-02-09 by AlexandreRouma

L1049G, 2023-07-18:
I am getting exactly the same error;
$ sdrpp
sdrpp: error while loading shared libraries: libvolk.so.2.2: cannot open shared object file: No such file or directory

Using the nightly jammy deb; sdrpp_ubuntu_jammy_amd64.deb
on Ubuntu 22.04 (jammy).
libvolk.so.2.5 is installed