AlexandreRouma/SDRPlusPlus #990: Latest MacOS version competely broken

loewal, 2023-02-15:
**Hardware**
- CPU: iMac 21,5"
- RAM: 8Gb
- GPU: 
- SDR: SDRDX

**Software**
- Operating System: Ventura
- SDR++: compilation of #924

**Bug Description**
The app starts up, but is completely empty; without any module;
Previous version restored and is oke.

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.
[image]

AlexandreRouma, 2023-02-15:
924 is an experimental branch, it's not finished and won't work.
Never download any build that isn't from the `master` branch.
The correct place to download nightly builds is always https://www.sdrpp.org/nightly

closed 2023-02-15 by AlexandreRouma

loewal, 2023-02-15:
Okay Alexandre,
Merci/Thanks!