AlexandreRouma/SDRPlusPlus #996: SoapySDR source missing from MacOS nightly build

yoramrotbach, 2023-02-24:
**Feature description**
Hi
Please add SoapySDR option to MacOS nightly build.

Thank you

Yoram

AlexandreRouma, 2023-02-24:
Not possible to do reliably.

Additionally, SoapySDR support is being phased out since basically all the hardware soapy supports is supported natively.

Not sure which hardware you're using specifically but there's very likely a native source for it.

closed 2023-02-24 by AlexandreRouma

yoramrotbach, 2023-02-24:
Hello
Thanks for the quick response.
I'm using CaribouLite SDR which currently is supported only by SoapySDR API.

Yoram

AlexandreRouma, 2023-02-24:
The cariboulite is a raspberry pi SDR... Why would you need soapy on the Mac to use it?
Just use SDR++ server on the pi and connect to it.

Moreover native support for the cariboulite is coming soon.

yoramrotbach, 2023-02-24:
Thank you
Will wait patiently
All the best
Yoram

AlexandreRouma, 2023-02-24:
There is no need to wait, Soapy is supported on the Pi. Just use SDR++ server and you'll see soapy in the remote source list.

Aldekein, 2023-06-08:
Hey, I'm trying to use SoapySDR in a Mac build to connect to and SDRPlay compatible SDR Receiver RSp1 with MSi001 and MSi2500 Main Chips. It's a pity to see it's being removed.

aneitzel, 2024-09-24:
Same here - with removal of SOAPY my SDRplay compatible SDR dont work anymore. SDRplay dont support my "compatibel" device, so SDRplay native support is not an option. 

**Is there any way to have it BACK in SDR++? Its sad to remove things which worked good so far.**