AlexandreRouma/SDRPlusPlus #1004: Ignore silence during recording

infrat, 2023-03-02:
**Feature description**
From what I see, in a recorder module you've decided to remove `ignoreSilence` option. 
https://github.com/AlexandreRouma/SDRPlusPlus/blob/master/misc_modules/recorder/src/main.cpp#L318
Are there any plans to revert it back? What are the issues here with this option?

Thank you @AlexandreRouma  for your hard work. 
Regards!

AlexandreRouma, 2023-03-02:
Added it back, it was removed because it didn't work properly and I didn't have time to fix it.

closed 2023-03-02 by AlexandreRouma

infrat, 2023-03-02:
Awesome, thx. I guess it works now?

infrat, 2023-03-03:
Yes, I can confirm to myself - I've jest downloaded the latest build and it works like a charm.

Btw. @AlexandreRouma do you consider introducing some additional features to this recorder? Like splitting records, or including delay after transmission, etc? 

Now I see a potential memory overflow because of having the active recordings stored in memory until user hit the record button. Assuming the user is recording multiple sinks (like me - I have 6 VFO and each I'm recording to a separate file), there might by significant amount of data in memory. Flushing them to disk from time to time, might be a good idea.