AlexandreRouma/SDRPlusPlus #1011: WSJTX using RIGCTL server crashes

ja-Bertolin, 2023-03-13:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU:  2.6 GHz Quad-Core Intel Core i7
- RAM:16 GB 2133 MHz LPDDR3
- GPU: Radeon Pro 450 2 GB
- SDR: local

**Software**
- Operating System: MacOS 11.7.4
- SDR++: Version 1.1.0 + Build date 17:21:28, Mar 13 2023

**Bug Description**
I want to connect SDR++ with WSJTX for receiving WSPR and once configured it fails giving the following error

Hamlib error: rig.c(1807) trace
netrigctl_set_freq called
netrigctl_vfostr: called vfo=currVFO
netrigctl_vfostr: vfo==RIG_VFO_CURR, curr=VFO
netrigctl_vfostr: vfo_opt=1
netrigctl_vfostr: vfo_opt vfo=134217728
netrigctl_transaction: called len=23
rig_flush: called for network device
network_flush called
write_block(): TX 23 bytes, method=2
0000    46 20 56 46 4f 41 20 31 34 30 37 35 30 35 35 2e     F VFOA 14075055.
0010    30 30 30 30 30 30 0a                                000000.         
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): RX 7 characters, direct=1
0000    52 50 52 54 20 31 0a                                RPRT 1.         
netrigctl_set_freq: cmd=F VFOA 14075055.000000
0:rig.c(1814):rig_set_freq returning(-8) Protocol error

Protocol error
Protocol error
 while setting frequency

**Steps To Reproduce**
1. In SDR++ Start Rigctl Server at 127.0.0.1: 4532
2. In WSJTX I configure 
3. 
[image]

4. When press OK I receive the message and 
[image]

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

ja-Bertolin, 2023-03-13:
By the way, just in case it helps,  I have tried with GQRX and it works like a charm. Following the same steps.

AlexandreRouma, 2023-03-13:
Duplicate of https://github.com/AlexandreRouma/SDRPlusPlus/issues/573

closed 2023-03-13 by AlexandreRouma