AlexandreRouma/SDRPlusPlus #1014: SoapySDR Remote source

great9, 2023-03-16:
**Feature description**
Any plans to add SoapySDR Remote as a source?

AlexandreRouma, 2023-03-16:
1) SoapySDR is supported, any soapy server on your network will show up
2) Using SoapyRemote is dumb. Use SDR++ server instead.

closed 2023-03-16 by AlexandreRouma

Aldekein, 2023-06-08:
Hey @AlexandreRouma 

So I run:

```# SoapySDRServer --bind 0.0.0.0

######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.6.0-gf375555e
Server UUID: 05a13ccf-0378-1000-81a7-23b400000000
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server...
Connecting to DNS-SD daemon...
[INFO] mDNSResponder version: v180710.10.2
[INFO] Bonjour daemon version: v180712.10.1
[INFO] DNSServiceRegister(M2\032Space\032Apple._soapy._tcp.local.)
Press Ctrl+C to stop the server```

But I don't see any way to connect to this source in SDR++ :(