AlexandreRouma/SDRPlusPlus #1020: Raspberry Pi OpenGL issue

KC9RFV, 2023-03-22:
I am trying to install SDRPlusPlus on my Raspberry Pi, but am coming up with a loss. I am still getting familiar with Linux and Raspberry Pi. Please be descriptive with any resolutions. I have been pulling my hair out (If I had any) trying to figure this out...

Raspberry Pi 3b (no +)

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian

SDRPP nightly build - Nov 22, 2022

The software installs with no issues. But when I try to run the program from terminal I get the following errors and SDR++ closes:

****
[22/03/2023 10:17:20.000] [INFO] SDR++ v1.1.0
[22/03/2023 10:17:20.000] [INFO] Loading config
[22/03/2023 10:17:20.000] [WARN] ConfigManager locked, waiting...
[22/03/2023 10:17:20.000] [ERROR] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
[22/03/2023 10:17:20.000] [INFO] OpenGL 3.0 was not supported
[22/03/2023 10:17:20.000] [ERROR] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
[22/03/2023 10:17:20.000] [INFO] OpenGL 3.1 ES was not supported
[22/03/2023 10:17:20.000] [INFO] Using OpenGL 2.1
Failed to initialize OpenGL loader!
[22/03/2023 10:17:21.000] [WARN] Could not init using OpenGL with normal GLSL version, falling back to GLSL 1.2
Failed to initialize OpenGL loader!
[22/03/2023 10:17:21.000] [ERROR] Failed to initialize OpenGL with GLSL 1.2
****

I have already opened raspi-config --> advanced options --> GL Driver --> Full KMS. 

That has not seemed to work for me. I am good at doing research but this is either a new issue, or something alot of people have not had issues with. Any help on this would be appreciated. I am SO at a loss on this one...

KC9RFV, 2023-03-22:
OMG IT WORKS!!! 

Thank you so much for your input! 

I'm not sure if you can, but could you explain what the above commands do? I have been going crazy trying to get SDR++ to run. I've been working on it for over 2 weeks researching and trying different ways, and this seems so simple.

Thanks Again!
73, David

ea4bpn, 2023-03-23:
Glad to know it is working now !

Those commands modify an environment variable (similar to MSwindows). The first one adds a value to the MESA_GL_VERSION_OVERRIDE environment. Second one export such variable.

Take in mind you need to set these commands everytime you start session on your raspberry; they are not persistent between sessions. You may add it to .bashrc file but for the moment I do not recommend it. Perhaps the author is aware of this and he will know how to solve that more efficiently.

For the error itself, I do not know why it happen. The link in previous message redirect to the minecraft forums and there is not a clear answer.

Enjoy sdr++  
73, Rafael

KC9RFV, 2023-03-23:
Thank you for the explanation. I created a .sh on my desktop and made it executable to run these commands and then start SDR++. Works perfect... 
Thanks for the help...

closed 2023-03-23 by KC9RFV

Johng117, 2024-01-29:
Thanks for this fix, I can now get raylib to work on my pi.

OctaAIVision, 2024-11-28:
@ea4bpn 
thanks man u saved my life. <3