AlexandreRouma/SDRPlusPlus #1023: Radio Check Mark closes program, and audio only shows headphones....

KC9RFV, 2023-03-23:
Ok, I have an interesting one. 

At some some point I had unchecked the checkbox beside the radio menu. Now, when I try to recheck the box, the program closes. I have tried a different SDR Is there someway to delete the old settings and start fresh? Is the a way to delete the old settings file or folder? I have tried uninstalling and reinstalling with no help. There are too many folders to try and track them down. 

Next issues is under audio sink, it only shows:

 hw:bcm2835 Headphones,0

I am using a HDMI monitor that has speakers. Everything else on raspiberry pi uses those. I cannot get the speakers to recognize through SDR++. Anyone else had an issue with this?

Raspberry Pi 3b (no +)

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian

SDR++ Nightly build Nov 22, 2022

ea4bpn, 2023-03-25:
> Is there someway to delete the old settings and start fresh? Is the a way to delete the old settings file or folder?

The settings is in your home directory:
~/.config/sdrpp

Regarding raspberry pi, it seems to me it is not powerful enough for sdr++
Instead, I would run sdr++ as a server and then connect it from any other PC. Check page 58 on the well written manual.

AlexandreRouma, 2023-09-13:
Same root issue as #816, follow that one instead.

closed 2023-09-13 by AlexandreRouma