AlexandreRouma/SDRPlusPlus #1025: LimeSDR Mini v2 does not immediately start reception

sergeyvfx, 2023-03-24:
**Hardware**
- CPU: 2,7 GHz Quad-Core Intel Core i7
- RAM: 16GB
- GPU: Intel Iris Plus Graphics 655 1536 MB
- SDR: LimeSDR Mini v2, USB

**Software**
- Operating System: macOS Ventura 13.2.1 (22D68)
- SDR++: Built from Git hash ec4dc6cc9e9 (nightly builds are not suitable, as a newer version of LimeSuite library is needed)

**Bug Description**
Starting the radio (clicking on the "Start" button) does not immediately start reception. A workaround from the interface is possible: change the bandwidth.

**Steps To Reproduce**
1. Plug LimeSDR Mini v2
2. Launch SDR++
3. Click "Start" button
4. Observe that nothing seems to happen
5. Change bandwidth
6. The reception (audio, FFT updates, etc) starts

**Additional info**
Some observations which might help with troubleshooting and finding proper solution:
- Calling `LMS_SetLPFBW` right after the `LMS_StartStream` makes the reception start as expected
- The suggestion on the [LimeSDR forum](https://discourse.myriadrf.org/t/limesdr-mini-v2-reading-samples-from-rx-fifo/7900) is to use `LMS_EnableChannel` on the TX channel (which matches the demo code)

SimonG4ELI, 2023-04-08:
After starting streaming data, run the RX Calibration. This then ensures data streaming.

sergeyvfx, 2023-04-08:
@SimonG4ELI This is good to know! I can confirm it does solve the problem. It is just a bit strange since typically it seems calibration is done before starting the stream. 

Anyhow, I'll add this observation to the initial post so that it is not lost :)

SimonG4ELI, 2023-04-08:
> @SimonG4ELI This is good to know! I can confirm it does solve the problem. It is just a bit strange since typically it seems calibration is done before starting the stream.
> 
> Anyhow, I'll add this observation to the initial post so that it is not lost :)

Someone told me about this - it's documented in a library update but I also never saw it!

AlexandreRouma, 2023-10-30:
I'm pretty sure this was just a bug early after release, reopen if this still happens with the latest version of limesuite.

closed 2023-10-30 by AlexandreRouma

sergeyvfx, 2023-12-09:
Think if that was considered a bug in the LimeSuite it would have been acknowledged by the LimeSDR team. But they explicitly state that both TX and RX channels are to be enabled due to the gateware implementation.

Nevertheless, I did give it a test with the latest LimeSuite `c2d9e877545` and the SDR++ 794d6ff5acf. The issue is still present. I've tested with self-compiled SDR++ as the nightly builds as they are using old LimeSuite which does not support LimeSDR 2.

I can not re-open the issue, but it really is not resolved.

AlexandreRouma, 2023-12-09:
Definitely a bug on their end. If their new gateware requires nonsensical configuration to work as espected, they should either fix it or work around it in limesuite.

sergeyvfx, 2023-12-09:
The bottom line is: this bug is not fixed.

AlexandreRouma, 2023-12-09:
Then report it to LimeSuite, not to me.