AlexandreRouma/SDRPlusPlus #1026: undefined reference to wl*

4e696e6a6121, 2023-03-26:
I am trying to compile SDRPlusPlus on Fedora 37 but I am failing at the 'make' stage. I have attached files that show what happens at each stage.

Cmake seems to check out fine with all the required dependencies being met. Running make straight after goes up to 64% and then fails.

I did a bit of research into what all the wl_* references were and it seems to be pointing towards wayland. I am running the KDE Plasma interface so not sure if that has anything to do with it.

There is a sddr++ package on the dnf repositories but there seems to be an issue with no audio being heard. So I took the route of just manually compiling rtaudio as there was a reference to the specific issue I was coming across. 

Any pointers or suggestions as to how I can overcome this error would be much appreciated. Please let me know if you require more information. The output of the `uname -r` command is:

`6.2.7-200.fc37.x86_64`

[make.log](https://github.com/AlexandreRouma/SDRPlusPlus/files/11073208/make.log)
[error.log](https://github.com/AlexandreRouma/SDRPlusPlus/files/11073209/error.log)
[cmake.log](https://github.com/AlexandreRouma/SDRPlusPlus/files/11073210/cmake.log)

AlexandreRouma, 2023-03-26:
You probably installed the wrong GLFW for your fedora install. There's a wayland one and a x11 one.

4e696e6a6121, 2023-03-27:
Hi @AlexandreRouma ,

Many thanks. I removed the glfw program completely and re-compiled from source again. This time checking that I did type in the below:

`cmake -S . -B build -D GLFW_USE_WAYLAND=1`

The build above goes through ok and cmake from SDRPlusPlus detects GLFW fine. But running make produces the same results.

AlexandreRouma, 2023-03-30:
I don't know anything about building GLFW from source and I don't use wayland or fedora either, just install the correct one from your distro's package manager. All I know is it works fine for other people using wayland so it must be something about your specific install. Suggest you open an issue for the GLFW developers instead.

closed 2023-03-30 by AlexandreRouma