AlexandreRouma/SDRPlusPlus #1031: CAT communication and "IF listening" mode in ham radio

canislupus11, 2023-03-30:
Ability to communicate with an amateur radio via CAT (rigctl?), especially in "IF listening" mode when the SDR is tuned to a fixed frequency, but clicking on the waterfall will tune the radio according to the distance on the waterfall and taking into account the currently tuned frequency reads from the radio after the CAT .

AlexandreRouma, 2023-03-30:
the rigctl server is already fully implemented and the rigctl client is partially implemented (duplicate of https://github.com/AlexandreRouma/SDRPlusPlus/issues/174)

closed 2023-03-30 by AlexandreRouma

canislupus11, 2023-03-30:
> the rigctl server is already fully implemented and the rigctl client is partially implemented (duplicate of #174)

What about "IF listening" mode?

AlexandreRouma, 2023-03-30:
that's what the rigctl client does