AlexandreRouma/SDRPlusPlus #1032: FFT Smoothing

kmp56, 2023-03-30:
**Hardware**
- CPU: i9 12900K
- RAM: 32GB
- GPU: nVidia GTX1650
- SDR: SDRPlay RSP1A. AirSpy-mini (both USB)

**Software**
- Operating System: Windows 11 x64, Ubuntu 22.04
- SDR++: Version 1.1.0,  build 30.03.2023

**Bug Description**
The new FFT Smoothing setting works very nicely, but when configured to auto start it does not start with SDR++. Only entering the settings and switching OFF / ON again makes it work properly. The problem occurs in Linux (Ubuntu 22.04) and Windows11Pro.

AlexandreRouma, 2023-03-30:
should be fixed, wait 30 minutes for the nightly to finish building and check if the issue is resolved.

kmp56, 2023-03-30:
Excellent ! I have already compiled the Linux version (github build 800a8b2) - it works as expected.
Thanks for your quick response!
Regards
kmp56

closed 2023-03-30 by kmp56