AlexandreRouma/SDRPlusPlus #1042: RTL-SDR could not connect to Android version of SDR++

Pravdyvy, 2023-04-10:
**Hardware**
- CPU: Qualcomm Snapdragon 720G 2.32 GHz 
- RAM: 6 GB
- GPU: Internal
- SDR: local RTL-SDR

**Software**
- Operating System: Android 10
- SDR++: nightly build; date : April 3, 2023

**Bug Description**
RTL-SDR could not connect to Android version of SDR++ .

**Steps To Reproduce**
1. Plug USB-c to USB 2.0 connector
2. Plug RTL-SDR
3. Start SDR++
5. Push Refresh
6. Observe RTL-SDR Dongle USB at second line of Source module
7. Pushing start does not start waterfall

**Screenshot**
https://drive.google.com/file/d/1Zo7uDr3Po4wa421PNORLbawaAjFpKpbH/view?usp=share_link

**Additional info**
Gain field in Source module shows 0.0dB and slider is as long as whole field.

AlexandreRouma, 2023-04-10:
this is usually a sign of a bad USB cable or your phone not providing enough power for the dongle.

AlexandreRouma, 2023-04-23:
closing for unactivity

closed 2023-04-23 by AlexandreRouma