AlexandreRouma/SDRPlusPlus #1046: RigCTL Client status

ON4BCB, 2023-04-17:
**Feature description**
Hi,

is there any progres in the rigctl client module

I want to connect to a spyserver on the remote site connected as panadapter to a normal radio
the sdr++ will be used remotly and control/follow the remote radio by CAT

Looking forward to test

darksidelemm, 2023-05-06:
I had a quick go at this with my IC-7610. Some notes:
- I was using WFview to present a rigctld server for my IC-7610, which it was connected to via network.
- Rigctld Client had to be enabled from the Module Manager.
- It wasn't obvious to me, but it turns out you have to click the left/right arrow button to the right of the frequency entry to have the rig be commanded. There's no tooltip or any other indication as to what this button does (maybe I missed some form of documentation on this somewhere?)
- Control appears to be unidirectional (SDR++ to the rig you are connected to).
- Bi-directional control (probably via polling the rig at regular intervals) would be appreciated, so i can spin the dial on my rig, and have SDR++ follow along - this would be my typical use-case for this feature.

AlexandreRouma, 2023-10-30:
Duplicate, follow #174 instead.

closed 2023-10-30 by AlexandreRouma