AlexandreRouma/SDRPlusPlus #1052: No sound

shadyslim2018, 2023-04-22:
OS: Manjaro Linux x86_64 
Host: X570 AORUS MASTER -CF 
Kernel: 6.1.23-1-MANJARO 
DE: GNOME 43.4 
CPU: AMD Ryzen 9 5950X (32) @ 3.400GHz 
GPU: AMD ATI Radeon RX 6700/6700 XT/6750 XT / 6800M/6850M XT 
SDR RSPdx local

No sound at all, and the radio has been ticked. The sound works fine in CubicSDR. I have tried the nightly build also, but it didn't work.

AlexandreRouma, 2023-04-22:
No logs, no screenshot of the sound menu, no steps to reproduce.
How exactly am I supposed to be able to help with anything here?

shadyslim2018, 2023-04-22:
Sorry for not providing more info.

There is no special steps. Just run the software and there is just no sound, you can see the graph move in the recorder module.
```
[INFO] SDR++ v1.1.0
[INFO] Loading config
[INFO] Using OpenGL 3.0
[ERROR] Glfw Error 65544: Wayland: Setting window icon not supported
[INFO] Loading icons
[INFO] Loading band plans
[INFO] Loading band plans color table
[INFO] Loading modules
[INFO] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/audio_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/file_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[INFO] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[INFO] Loading /usr/lib/sdrpp/plugins/network_sink.so
[INFO] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[INFO] Loading /usr/lib/sdrpp/plugins/radio.so
[INFO] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[INFO] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[INFO] Loading /usr/lib/sdrpp/plugins/recorder.so
[INFO] Loading /usr/lib/sdrpp/plugins/rigctl_client.so
[INFO] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[INFO] Loading /usr/lib/sdrpp/plugins/scanner.so
[INFO] Initializing Airspy Source (airspy_source)
[INFO] Initializing AirspyHF+ Source (airspyhf_source)
[INFO] Initializing Audio Sink (audio_sink)
[INFO] Initializing Audio Source (audio_source)
[INFO] New DSP samplerate: 48000.000000 (source samplerate is 48000.000000)
[INFO] Initializing BladeRF Source (bladerf_source)
[ERROR] Module 'bladerf_source' doesn't exist
[INFO] Initializing File Source (file_source)
[INFO] Initializing Frequency Manager (frequency_manager)
[INFO] Initializing HackRF Source (hackrf_source)
[INFO] Initializing Hermes Source (hermes_source)
[INFO] Initializing LimeSDR Source (limesdr_source)
[ERROR] Module 'limesdr_source' doesn't exist
[INFO] Initializing Network Sink (network_sink)
[INFO] Initializing PlutoSDR Source (plutosdr_source)
[INFO] Initializing RFspace Source (rfspace_source)
[INFO] Initializing RTL-SDR Source (rtl_sdr_source)
[INFO] Initializing RTL-TCP Source (rtl_tcp_source)
[INFO] Initializing Radio (radio)
[Resamp] predec: 8192, interp: 25, decim: 6, inacc: 0.000000%, taps: 1900
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[Resamp] predec: 32, interp: 5000, decim: 7813, inacc: 0.000000%, taps: 593750
[Resamp] predec: 8192, interp: 125, decim: 24, inacc: 0.000000%, taps: 9500
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[WARN] Demod switch took 26040 us
[INFO] RtAudio stream open
[INFO] Initializing Recorder (recorder)
[INFO] Initializing Rigctl Server (rigctl_server)
[INFO] Initializing SDR++ Server Source (sdrpp_server_source)
[INFO] Initializing SDRplay Source (sdrplay_source)
[ERROR] Module 'sdrplay_source' doesn't exist
[INFO] Initializing SoapySDR Source (soapy_source)
CPU Features: SSE2+ SSE4.1+ AVX+ FMA+
[... 51 lines cut ...]
[INFO] Running post-init for RFspace Source
[INFO] Running post-init for RTL-SDR Source
[INFO] Running post-init for RTL-TCP Source
[INFO] Running post-init for Radio
[INFO] Running post-init for Recorder
[INFO] Running post-init for Rigctl Server
[INFO] Running post-init for SDR++ Server Source
[INFO] Running post-init for SoapySDR Source
[INFO] Running post-init for SpyServer Source
[INFO] Ready.
[INFO] devIdx: 0
[INFO] SerNo: 22400CC748
[INFO] hwVer: 4
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000
[INFO] Bandwidth for samplerate 8000000.000000 is 8000000.000000
[INFO] Using format CF32.
[INFO] SoapyModule 'SoapySDR Source': Start!
[INFO] SoapyModule 'SoapySDR Source': Tune: 100261357.039187!
[INFO] SoapyModule 'SoapySDR Source': Stop!
```
[image]

[image]

AlexandreRouma, 2023-04-22:
Show a screenshot of the sinks menu. That's where the sound options are.

shadyslim2018, 2023-04-22:
It worked after a new Manjaro update with new kernel. The new kernel version is now 6.1.25-1-MANJARO.

Originally the sink menu didn't show my sound card, but after this update it has.

AlexandreRouma, 2023-04-22:
> It worked after a new Manjaro update with new kernel.

I'll assume this was a sound driver problem and close this issue.

closed 2023-04-22 by AlexandreRouma