AlexandreRouma/SDRPlusPlus #1058: Set current noise level as squelch value

sergeibit, 2023-05-08:
**Feature description**

Assuming there's no way to add a auto-squelch feature, it would be very helpful to have a button on the side of squelch setting to set the value to the current noise level.

[image]

AlexandreRouma, 2023-05-13:
Is anyone actually gonna use this?

It just as fast to just slide the slider until the audio blanks out continuously. Having an automated system to figure out the level at which you want it to open would probably not work as well as just doing that.

It's added complexity and compute power for something I don't think anyone really needs except the extremely inexperienced...

sergeibit, 2023-05-16:
Yes, I would use it. Remember: Just because you don´t need, doesn´t mean no one else need it. There are people with different objectives and needs (or maybe impairments? it doesn't matter) that would be positively affected with every small usability suggestion shared with you :) But reading other suggestions here I see there's a lot of resistance to implement any change to the software. I feel sad because SDR++ seems to have potential to become the "Go-To" software for SDR, but it feels it's just for the extremely experienced people. Maybe in the future. For now, it's better to close this suggestion and keep your roadmap as is :) Kind regards!