AlexandreRouma/SDRPlusPlus #1060: Bookmark sorting

sergeibit, 2023-05-08:
**Feature description**
I would like to suggest the ability to sort bookmarks by name or frequency, ascending or descending, by clicking on the column header.

[image]

It would also be very useful if the mode (NFM, WFM, AM, etc) had a separated column and could also be sorted.

sergeibit, 2023-05-16:
Reading the json format I saw they used the Name as key. That's just stupid in a basic software design level.
Feeling this is just another issue that will never be addressed and will just cause stress for nothing, I'm closing this suggestion.

AlexandreRouma, 2023-05-16:
If it wasn't gonna be addressed it would have been closed a long time ago.

You have to understand that I'm a university student and my free time is not infinitely extendable, especially a month away from the exams that'll determine if I graduate or not...

Daniel-from-Germany, 2023-07-21:
I hope you have already passed an exam and can now relax

maybe try this: *UNTESTED* !!! 

`// The loadByName function with the revised sorting function
void loadByName(std::string listName) {
    bookmarks.clear();
    if (std::find(listNames.begin(), listNames.end(), listName) == listNames.end()) {
        selectedListName = "";
        selectedListId = 0;
        loadFirst();
        return;
    }
    selectedListId = std::distance(listNames.begin(), std::find(listNames.begin(), listNames.end(), listName));
    selectedListName = listName;
    config.acquire();

    // Here we sort the channel names using the revised compare function
    std::vector<std::string> sortedChannelNames;
    for (auto [bmName, bm] : config.conf["lists"][listName]["bookmarks"].items()) {
        sortedChannelNames.push_back(bmName);
    }
    std::sort(sortedChannelNames.begin(), sortedChannelNames.end(), compareChannelNames);

    for (const auto& bmName : sortedChannelNames) {
        FrequencyBookmark fbm;
        fbm.frequency = config.conf["lists"][listName]["bookmarks"][bmName]["frequency"];
        fbm.bandwidth = config.conf["lists"][listName]["bookmarks"][bmName]["bandwidth"];
        fbm.mode = config.conf["lists"][listName]["bookmarks"][bmName]["mode"];
        fbm.selected = false;
        bookmarks[bmName] = fbm;
    }
    config.release();
}
`

rogerxxxx, 2024-03-29:
Better yet, most just simply desire sorting by frequency.

Sorting is already being completed by the first field, the "Name" of the bookmark, with a second field following titled "Bookmark" for the column containing frequencies.

I suggest merrily, switching the print order of the column field names to; first field "Bookmark" and second field "Name".

Then rename the "Bookmark" column title to "Frequency"; with sorting occurring by simple numerical order.

Users, if they desire another sorting order, can easily create a second or subsequent "List" (or group) name, augmenting which bookmarks/frequencies are display.

However, something crazy happens, sdrpp converts the frequency saved within the bookmark to common band plan frequency (eg. kHz for AM, mHz for FM), rather than conforming to one system of measurement.  So my suggestion will not work unless the frequency printed within the graphical interface uses either, all kHz or all mHz for every bookmark.  Could also work some code, for sorting based upon the hZ frequency saved within the bookmark file,  however this would consume more CPU resources.

Another workaround users can employ currently, prefixing their bookmark names with the actual frequency. (eg. all hZ, or all kHz, or all mHz.

One good note, sorting is occurring correctly for numerical ordering, at least here on my computer with sdpp instance running.  (eg. 0.01, 0.10, 1.0, ...)