AlexandreRouma/SDRPlusPlus #1061: RigCTL Server crashing episodically and cannot be restarted

mgiugliano, 2023-05-10:
**Hardware**
- CPU: 2.3 GHz 8-Core Intel Core i9
- RAM: 32 GB 2667 MHz DDR4
- GPU: AMD Radeon Pro 5500M 4 GB
- SDR: local, RSP1A

**Software**
- Operating System: MacOS 13.3.1 (a) (22E772610a)
- SDR++: Version 1.1.0 + Build date nightly build 19:26 May 2nd 2023.

**Bug Description**
The RigCTL Server crashes and can't be restarted by pressing the button that now reads "Start" (and no longer "stop").

**Steps To Reproduce**
The server starts correctly and allows me to get and change the frequency, keeping it in sync, by a simple script written in bash or python (https://gist.github.com/mgiugliano/bb3e24abc1e27527863236ee64344155). After a while, while reading the frequency of the SDR++, the RigCTL server crashes and cannot be restarted without closing and relaunching the app.

Note: I resorted using nc instead of the rigctl client (e.g. by rigctl -v -m 2 -r localhost:4534 f) because the SDR++ server returns a strange error: rig_open: error = localhost:4534 Invalid argument. If I use echo and nc there is no error.

AlexandreRouma, 2023-05-10:
Please provide SDR++ logs as requested in the issue template.

mgiugliano, 2023-05-11:
@AlexandreRouma Thank you for your patience with me. I have omitted it as I thought it is required "Only If SDR++ fails to lauch or the SDR fails to start".

However the command line option "-c" does not work (see below)

`> /Applications/SDR++.app/Contents/MacOS/sdrpp -c
[11/05/2023 18:29:59.000] [INFO] SDR++ v1.1.0
Unknown argument
-a --addr		Server mode address
   --autostart		Automatically start the SDR after loading
-h --help		Show help
-p --port		Server mode port
-r --root		Root directory, where all config files are stored
-s --server		Run in server mode
`

Without the "-c" option, below is what I got on the stdout. I managed to crash the server, while SDR++ still running and this is the error message printed on stdout each time I tried clicking on the button "Start".

`
[11/05/2023 18:38:27.000] [ERROR] Could not start rigctl server: std::exception
`

Below the full output.

`
/Applications/SDR++.app/Contents/MacOS/sdrpp
[11/05/2023 18:34:27.000] [INFO] SDR++ v1.1.0
[11/05/2023 18:34:27.000] [INFO] Loading config
[11/05/2023 18:34:27.000] [WARN] ConfigManager locked, waiting...
[11/05/2023 18:34:27.000] [INFO] Loading icons
[11/05/2023 18:34:27.000] [INFO] Loading band plans
[11/05/2023 18:34:27.000] [INFO] Loading band plans color table
[11/05/2023 18:34:27.000] [INFO] Loading modules
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rtl_sdr_source.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/meteor_demodulator.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/airspy_source.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/network_sink.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/scanner.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/plutosdr_source.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/radio.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rigctl_server.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/audio_sink.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/m17_decoder.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/frequency_manager.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/recorder.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/file_source.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/bladerf_source.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/airspyhf_source.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/spyserver_source.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rigctl_client.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/new_portaudio_sink.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/hermes_source.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/limesdr_source.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/discord_integration.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rfspace_source.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rtl_tcp_source.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/hackrf_source.dylib
[11/05/2023 18:34:27.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/sdrpp_server_source.dylib
[11/05/2023 18:34:27.000] [INFO] Initializing Airspy Source (airspy_source)
[11/05/2023 18:34:27.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[11/05/2023 18:34:27.000] [INFO] Initializing Audio Sink (audio_sink)
[11/05/2023 18:34:31.000] [INFO] Initializing Audio Source (audio_source)
[11/05/2023 18:34:31.000] [ERROR] Module 'audio_source' doesn't exist
[11/05/2023 18:34:31.000] [INFO] Initializing BladeRF Source (bladerf_source)
[11/05/2023 18:34:31.000] [ERROR] Could not list devices -7
[11/05/2023 18:34:31.000] [INFO] Initializing File Source (file_source)
[11/05/2023 18:34:31.000] [INFO] Initializing Frequency Manager (frequency_manager)
[11/05/2023 18:34:31.000] [INFO] Initializing HackRF Source (hackrf_source)
[11/05/2023 18:34:31.000] [INFO] Initializing Hermes Source (hermes_source)
[11/05/2023 18:34:31.000] [INFO] Initializing LimeSDR Source (limesdr_source)
[... 280 lines cut ...]
[11/05/2023 18:38:15.000] [INFO] Rigctl command: '\set_freq 7782450.0'
[11/05/2023 18:38:15.000] [INFO] Rigctl command: '\get_freq '
[11/05/2023 18:38:16.000] [INFO] Rigctl command: '\get_freq '
[11/05/2023 18:38:16.000] [INFO] Rigctl command: '\get_freq '
[... cut ...]
[11/05/2023 18:38:35.000] [ERROR] Could not start rigctl server: std::exception
[11/05/2023 18:38:35.000] [ERROR] Could not start rigctl server: std::exception
[11/05/2023 18:38:47.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Stop!
[11/05/2023 18:38:48.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Start!
[11/05/2023 18:38:48.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Tune: 8431684.221101!
2023-05-11 18:38:48.670 sdrpp[8607:375860] CoreText: System LastResort not available, using built-in copy.
[11/05/2023 18:38:50.000] [INFO] Exiting successfully
`

mgiugliano, 2023-05-12:
Thank you for the time you spent on it!

I resorted to "nc" netcast precisely because rigctl was NOT working. As I wrote in my first post here, 

> rigctl -v -m 2 -r localhost:4534 f

returned a strange error: rig_open: error = localhost:4534 Invalid argument. I am puzzled with it, as I a using rigctl with rigctld (controlling my USB transceiver without problems)

The fact that nc worked suggested that maybe there was something I did not understand on the syntax of rigctl.

Any advice from you would be precious.

AlexandreRouma, 2023-05-12:
The issue isn't with netcat. Calling the rigctl command repeatedly would result in the exact same issue. 
Just run netcat or rigctl continously and give it commands via it's stdin/stdout. (or better, just use python sockets).

For the rigctl error, it's because `rigctl` sends unsupported commands to SDR++ like `\get_powerstat` which are not supported because SDR++ only currently supports a subset of commands. It wasn't really meant to be used with hamlib directly, it was meant originally for software like gpredict that send commands directly themselves.

SO again, I recommend you use python sockets to issue commands manually instead.

mgiugliano, 2023-05-12:
BTW: I am not really sure the part of the code (you correctly criticised) is de facto executed 50 time/s, since the "echo | nc" has some lag to execute and return.

From the stdout it seems at most ~10 time/s when I kept acting on the (physical) VFO of my radio.

Thank you again for your time.

AlexandreRouma, 2023-05-12:
The delay in the code is set to 20ms = 50Hz. The additional delay caused by the commands is hardware specific.

lwvmobile, 2023-05-13:
Just as a casual aside, I've coded DSD-FME to make extensive use (abuse) of the RigCTL server and TCP Network Sink in SDR++, and depending on system type, I can hammer the RigCTL server with a get_freq request once every 80ms (NXDN SRV_INFO frame), send set_freq requests and tune around, and keep an incoming audio stream from TCP running all the same time, and SDR++ rock solid, no issues, whatsoever. On one of my remote locations, its been running for weeks at a time without issue and without disconnect, I often forget its running at all.

AlexandreRouma, 2023-05-13:
yes as long as you keep the connection continuous there is no limit to the frequency at which you can run commands.

mgiugliano, 2023-05-14:
So all in all, apologies for my inexperience. 

However, I did not get it: is it a bug or not that SDR++'s server can't be restarted after it fails?

mgiugliano, 2023-05-14:
@lwvmobile Thanks for the heads up! 

Do you think it might be sufficient for the case study of my own "sync" demo, to use/get inspiration from your "dsd_rigctl.c"?

lwvmobile, 2023-05-14:
> Do you think it might be sufficient for the case study of my own "sync" demo, to use/get inspiration from your "dsd_rigctl.c"?

If you want to, to be honest, a lot of what is in dsd_rigctl.c came from gqrx_scanner originally, but the general idea is just to connect to the TCP socket, and leave the connection open. Don't close the connection unless you close your software. Then, with it remaining open constantly, send whichever rigctl commands you want to (GetCurrentFreq or SetFreq, etc) and read the return.

mgiugliano, 2023-05-14:
Thank you @lwvmobile and @AlexandreRouma for guidance and patience with my ignorance.

For posterity, here is a working solution (written in C) that does NOT crash SDR++ server - https://github.com/mgiugliano/syncSDRpp

AlexandreRouma, 2023-05-14:
I'll still leave this open in case the work around I was thinking of is valid

sq2mo, 2024-07-13:
The thread here seems to be close enough to ask a question:
Does rigctl server serve only one connection? My experience is:
while one client is connected, subsequent clients are not being served nor replied.

AlexandreRouma, 2024-07-14:
the server is single client only

sq2mo, 2024-07-14:
Any plans to fix this in the nearest future?

AlexandreRouma, 2024-07-14:
no