AlexandreRouma/SDRPlusPlus #1062: Build for ARM64 on MacOS M1 or M2

bmentink, 2023-05-12:
Hi Please consider doing a build for ARM64, a lot of folk have MacOS M1 or M2 machines ..

Many Thanks

bmentink, 2023-05-12:
Tried to build for arm64. After installing all dependancies using brew got this on make:
```
  "_volk_malloc", referenced from:
      dsp::stream<dsp::complex_t>::stream() in gui.cpp.o
      dsp::stream<dsp::complex_t>::setBufferSize(int) in gui.cpp.o
      dsp::stream<unsigned char>::stream() in server.cpp.o
      dsp::stream<unsigned char>::setBufferSize(int) in server.cpp.o
      dsp::stream<dsp::complex_t>::stream() in server.cpp.o
      dsp::stream<dsp::complex_t>::setBufferSize(int) in server.cpp.o
      IQFrontEnd::init(dsp::stream<dsp::complex_t>*, double, bool, int, bool, int, double, IQFrontEnd::FFTWindow, float* (*)(void*), void (*)(void*), void*) in iq_frontend.cpp.o
      ...
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [core/libsdrpp_core.dylib] Error 1
make[1]: *** [core/CMakeFiles/sdrpp_core.dir/all] Error 2
make: *** [all] Error 2
```

AlexandreRouma, 2023-05-12:
> Hi Please consider doing a build for ARM64, a lot of folk have MacOS M1 or M2 machines

GitHub doesn't give any CI instances for arm64 and cross-compilation is not possible using brew (as already explained in this issue https://github.com/AlexandreRouma/SDRPlusPlus/issues/557).

> Tried to build for arm64. After installing all dependancies using brew got this on make:

I'm not sure how you expect anyone to be able to help with 10 lines of error out of an entire build log and no info on what command(s) you ran or even what OS and architecture this is running on

bmentink, 2023-05-13:
Dude your needs adjusting. This is not a bug report, it's a feature request!  Even if it was a bug report, with attitudes like that, why would someone want to help for free? I did mention the OS (Macos) and the architecture (M1,M2) ........

AlexandreRouma, 2023-05-13:
A feature request with a compile issue description sounds a lot like a malformatted bug report...
I don't expect any help for free. And "macOS" alone without a version means nothing and "M1/M2" doesn't either considering the error you posted says you're compiling on x86 and there was no way to know if you were cross compiling... Maybe if you did a search for other issues and realized that apple silicon builds are already supported which turns this into a bug report there would be no ambiguity.

closed 2023-05-13 by AlexandreRouma