AlexandreRouma/SDRPlusPlus #1069: Pipewire support

iamkarlson, 2023-05-15:
This is an ongoing issue that SDR++ failing to install with the RtAudio. I found the following issues that are trying to solve the problem 
* https://github.com/AlexandreRouma/SDRPlusPlus/issues/1056
* https://github.com/AlexandreRouma/SDRPlusPlus/issues/878
* https://github.com/AlexandreRouma/SDRPlusPlus/issues/734
* https://github.com/AlexandreRouma/SDRPlusPlus/issues/173

That always referring to the thread in https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112 to recompile SDR++ with Pulse (?). However, these changes aren't easy to apply if you aren't C++ dev. Is it by chance possible to make type of the sink to use parametrizable? so we can pass it as an argument during runtime and build a binary both for legacy audio systems and pipewire that seem to being adopted almost everywhere nowadays

AlexandreRouma, 2023-05-15:
> That always referring to the thread in https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112 to recompile SDR++ with Pulse (?).

I'm not sure if you've read those instructions at all but nowhere does it say you need to recompile SDR++.

> However, these changes aren't easy to apply if you aren't C++ dev. Is it by chance possible to make type of the sink to use parametrizable?

There is no C++ to edit at all, you just rebuild **RtAudio** with an additional command line flag in the build command.  This takes literally 2 commands...

> pipewire that seem to being adopted almost everywhere nowadays.

Pipewire emulates PulseAudio automatically, this is simply an issue with RtAudio's pulse implementation. Nothing wrong with either Pipewire or SDR++.

> Perhaps since AUR package is taken under control by the community after the orphanage [1], it's a good idea to restore arch support?

The AUR admins took the package away from me, as such, they can fuck off and I will not help in any way shape or form to maintain it (aka, do not expect support regarding packaging issues).
I recommend you instead install SDR++ from source which just requires a few commands listed in the readme.

closed 2023-05-15 by AlexandreRouma

AlexandreRouma, 2023-05-16:
You have no idea what you're talking about. They forcefully took the package away from me because I didn't change a minor versioning issue (with no visible problem for 99% of users) while I was dealing with serious problems in my family and you expect me to keep supporting these people? No.

Buttars, 2023-05-20:
They took away your package because when asked to resolve an extremely simple problem with the package, which is your duty as the package maintainer, you lashed out and attacked others personally. Additionally, you admitted in an extremely insulting and childish way that you did not have the time or resources to maintain the package. You refused any co-authors and showed no intention of resolving the issue in a reasonable time.

Why should they have kept you as the maintainer if you're both not maintaining the package and causing unnecessary drama?

The AUR admins had two choices, to remove you as the package maintainer because you weren't actively maintaining the package or let the package go stale until you felt like maintaining it.

You responding to these actions by being spiteful and angry isn't doing you any favors.

> I'm dealing with unversity and the death of a close family member, your
> fucking version issue is the least of my priority.

https://lists.archlinux.org/archives/list/aur-requests@lists.archlinux.org/thread/UA6AZJ3Z2OMAVXSV7KGEG573EKV5NPCS/#ZM7JB2VQX3WPTBDI6KCU5WY3REJPF37M

AlexandreRouma, 2023-05-20:
> you admitted in an extremely insulting

Only valid way to respond when someone acts like an entitled cunt **after** having been told on other platforms the reason for the issue not being fixed.

> and causing unnecessary drama?

Merely responding to someone's blatant entitlement is not causing drama. Going crying to the AUR devs after you were nicely told that the versioning issue was not a problem worth my time at the moment **is**.

> let the package go stale until you felt like maintaining it.

Bingo! None of their business how I manage my package when it had nothing to do with being stale, it just had a meaningless versioning issue (which I'll add was still valid in terms of `-git` packaging standards).

> I'm afraid your attitude towards others will repel other talented developers who wish to collaborate with you.

I have never asked for anyone's contributions. I don't make my projects open-source expecting collaboration, I make them open source for basic user freedom. I'm more than happy to NOT collaborate with people who think their little versioning issue affecting basically no one is worth wasting someone's time during some of the most difficult months of their life.