AlexandreRouma/SDRPlusPlus #1072: SDRpp segvfault with Hack RF One.

John-Telek, 2023-05-17:
Debian bullseye, I7 5820K 32GB 980GTX
Linux debian 5.10.0-22-amd64 #1 SMP Debian 5.10.178-3 (2023-04-22) x86_64 GNU/Linux

App segfaults upon activation or refresh of Hack RF.

[17/05/2023 16:21:46.000] [INFO] SDR++ v1.1.0
[17/05/2023 16:21:46.000] [INFO] Loading config
[17/05/2023 16:21:46.000] [INFO] Using OpenGL 3.0
[17/05/2023 16:21:46.000] [INFO] Loading icons
[17/05/2023 16:21:46.000] [INFO] Loading band plans
[17/05/2023 16:21:46.000] [INFO] Loading band plans color table
[17/05/2023 16:21:46.000] [INFO] Loading modules
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[17/05/2023 16:21:46.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so.
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/file_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/radio.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/m17_decoder.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_client.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/scanner.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_sink.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/bladerf_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/recorder.so
[17/05/2023 16:21:46.000] [WARN] ConfigManager locked, waiting...
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[17/05/2023 16:21:46.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[17/05/2023 16:21:46.000] [INFO] Initializing Airspy Source (airspy_source)
[17/05/2023 16:21:46.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[17/05/2023 16:21:46.000] [INFO] Initializing Audio Sink (audio_sink)
[17/05/2023 16:21:46.000] [INFO] Initializing Audio Source (audio_source)
[17/05/2023 16:21:46.000] [INFO] New DSP samplerate: 48000.000000 (source samplerate is 48000.000000)
[17/05/2023 16:21:46.000] [INFO] Initializing BladeRF Source (bladerf_source)
[17/05/2023 16:21:46.000] [ERROR] Could not list devices -7
[17/05/2023 16:21:46.000] [INFO] Initializing File Source (file_source)
[17/05/2023 16:21:46.000] [INFO] Initializing Frequency Manager (frequency_manager)
[17/05/2023 16:21:46.000] [INFO] Initializing HackRF Source (hackrf_source)
[17/05/2023 16:21:46.000] [INFO] Initializing Hermes Source (hermes_source)
[17/05/2023 16:21:46.000] [INFO] Initializing LimeSDR Source (limesdr_source)
[17/05/2023 16:21:46.000] [INFO] Initializing Network Sink (network_sink)
[17/05/2023 16:21:46.000] [INFO] Initializing PlutoSDR Source (plutosdr_source)
[17/05/2023 16:21:46.000] [INFO] Initializing RFspace Source (rfspace_source)
[17/05/2023 16:21:46.000] [INFO] Initializing RTL-SDR Source (rtl_sdr_source)
usb_claim_interface error -6
[17/05/2023 16:21:46.000] [ERROR] Could not open RTL-SDR: -6
[17/05/2023 16:21:46.000] [INFO] Initializing RTL-TCP Source (rtl_tcp_source)
[17/05/2023 16:21:46.000] [INFO] Initializing Radio (radio)
[Resamp] predec: 8192, interp: 25, decim: 6, inacc: 0.000000%, taps: 1900
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[Resamp] predec: 2, interp: 5, decim: 8, inacc: 0.000000%, taps: 608
[Resamp] predec: 8192, interp: 16, decim: 5, inacc: 0.000000%, taps: 1216
[Resamp] predec: 8192, interp: 16, decim: 5, inacc: 0.000000%, taps: 1216
[17/05/2023 16:21:46.000] [WARN] Demod switch took 920 us
[17/05/2023 16:21:46.000] [INFO] RtAudio stream open
[... 34 lines cut ...]
[17/05/2023 16:21:47.000] [INFO] Running post-init for AirspyHF+ Source
[17/05/2023 16:21:47.000] [INFO] Running post-init for Audio Sink
[17/05/2023 16:21:47.000] [INFO] Running post-init for Audio Source
[17/05/2023 16:21:47.000] [INFO] Running post-init for BladeRF Source
[17/05/2023 16:21:47.000] [INFO] Running post-init for File Source
[17/05/2023 16:21:47.000] [INFO] Running post-init for Frequency Manager
[17/05/2023 16:21:47.000] [INFO] Running post-init for HackRF Source
[17/05/2023 16:21:47.000] [INFO] Running post-init for Hermes Source
[17/05/2023 16:21:47.000] [INFO] Running post-init for LimeSDR Source
[17/05/2023 16:21:47.000] [INFO] Running post-init for Network Sink
[17/05/2023 16:21:47.000] [INFO] Running post-init for PlutoSDR Source
[17/05/2023 16:21:47.000] [INFO] Running post-init for RFspace Source
[17/05/2023 16:21:47.000] [INFO] Running post-init for RTL-SDR Source
[17/05/2023 16:21:47.000] [INFO] Running post-init for RTL-TCP Source
[17/05/2023 16:21:47.000] [INFO] Running post-init for Radio
[17/05/2023 16:21:47.000] [INFO] Running post-init for Recorder
[17/05/2023 16:21:47.000] [INFO] Running post-init for Rigctl Server
[17/05/2023 16:21:47.000] [INFO] Running post-init for SDR++ Server Source
[17/05/2023 16:21:47.000] [INFO] Running post-init for SoapySDR Source
[17/05/2023 16:21:47.000] [INFO] Running post-init for SpyServer Source
[17/05/2023 16:21:47.000] [INFO] Ready.
[17/05/2023 16:21:47.000] [INFO] HackRFSourceModule 'HackRF Source': Tune: 127446000.000000!
Segmentation fault

JT.

AlexandreRouma, 2023-05-17:
libhackrf bug.
This part of the troubleshooting section of the readme gives the solution https://github.com/AlexandreRouma/SDRPlusPlus#sdr-crashes-when-starting-a-hackrf

John-Telek, 2023-05-17:
Sorted. Thanks.

It appears though that this clone is deaf and google shows a lot of
complaints and issues along these lines.

And so far have had no luck with any app when it comes to what should
the gain/vga be set to and are there hidden or other controls for the
unit ?
The lna gain isnt sequential 1,2,3 but jumps in 8db steps which can
screw things up and I cant find the RF AGC. (unless it doesn't have one
???)

Regards,
   John.

AlexandreRouma, 2023-05-17:
The HackRF doesn't have an AGC. Its LNA can also only be adjusted in 8dB steps.
Generally you adjust the LNA to right before the device starts clipping and then you add a bit of VGA gain if needed.
The HackRF in general is quite deaf, it was meant as a cheap way to do wideband SDR capture/replay, not as a sensitive receiver.