AlexandreRouma/SDRPlusPlus #1073: /usr/bin/ld: cannot find -lzstd

Mudary, 2023-05-18:
**Hardware**
- CPU: RPI 3+ 
- RAM:  1 gig
- GPU:  
- SDR: Locally Nooelec RTL-SDR

**Software**
- Operating System: Name + Exact version: Ubuntu 22.04 Bullseye LTS
- SDR++: Version + Build date: Nightly manually install by source

mudary2@raspberrypi:~ $ cd SDRPlusPlus
mudary2@raspberrypi:~/SDRPlusPlus $ cd build
mudary2@raspberrypi:~/SDRPlusPlus/build $ cmake ..
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mudary2/SDRPlusPlus/build
mudary2@raspberrypi:~/SDRPlusPlus/build $ make -j4
[ 77%] Built target correct-reed-solomon
[ 77%] Built target correct-convolutional
[ 77%] Built target error_sim_shim
[ 77%] Built target correct-h
[ 77%] Built target error_sim
[ 77%] Built target correct_static
[ 77%] Built target discord-rpc
[ 77%] Built target correct
[ 77%] Built target sdrpp_core
[ 77%] Built target audio_source
[ 77%] Built target airspyhf_source
[ 77%] Built target sdrpp
[ 77%] Built target airspy_source
[ 77%] Built target file_source
[ 77%] Built target hackrf_source
[ 77%] Built target plutosdr_source
[ 77%] Built target hermes_source
[ 77%] Built target rfspace_source
[ 77%] Built target rtl_sdr_source
[ 77%] Built target sdrpp_server_source
[ 77%] Built target rtl_tcp_source
[ 77%] Built target soapy_source
[ 77%] Linking CXX shared library spectran_http_source.so
[ 77%] Built target audio_sink
[ 77%] Built target spyserver_source
[ 77%] Built target network_sink
[ 77%] Built target meteor_demodulator
[ 77%] Built target radio
/usr/bin/ld: cannot find -lzstd
[ 77%] Built target discord_integration
[ 77%] Built target frequency_manager
[ 77%] Linking CXX shared library recorder.so
collect2: error: ld returned 1 exit status
make[2]: *** [source_modules/spectran_http_source/CMakeFiles/spectran_http_source.dir/build.make:126: source_modules/spectran_http_source/spectran_http_source.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:1851: source_modules/spectran_http_source/CMakeFiles/spectran_http_source.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 77%] Built target rigctl_client
[ 77%] Built target rigctl_server
/usr/bin/ld: cannot find -lzstd
collect2: error: ld returned 1 exit status
make[2]: *** [misc_modules/recorder/CMakeFiles/recorder.dir/build.make:111: misc_modules/recorder/recorder.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:2103: misc_modules/recorder/CMakeFiles/recorder.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
mudary2@raspberrypi:~/SDRPlusPlus/build $

AlexandreRouma, 2023-05-18:
Install the `libzstd-dev` package.

closed 2023-05-18 by AlexandreRouma

Mudary, 2023-05-18:
I did.
mudary2@raspberrypi:~/SDRPlusPlus/build $ sudo apt install libzstd-dev
[sudo] password for mudary2:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libzstd-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libzstd-dev' has no installation candidate

AlexandreRouma, 2023-05-18:
well you didn't because it's clearly telling you that it didn't install it.
I don't know what distro you're running but if it doesn't have libzstd it's probably time to update to something more recent.
No currently supported debian doesn't have that library.

Mudary, 2023-05-18:
Ok thanks, Great software on my other systems.