AlexandreRouma/SDRPlusPlus #1076: Two soundcards - two different outputs

brane0690, 2023-05-23:
Great work! Absolutely love SDR++

**Feature description**
It would be great to be able to send two different audios to two different sound cards simultaneously.
What I want to achieve is to send normal FM audio to sound card 1, and MPX FM audio to sound card 2.

And just a few other ideas... 
- ASIO audio output
- RDS decoding should be more sensitive - better threshold for lower RF reception
- dB signal power diagnostics
- SpyServer IP:PORT history (to be able to easily select between a few SpyServers)
- Volume numbers

Best regards.

AlexandreRouma, 2023-05-23:
> It would be great to be able to send two different audios to two different sound cards simultaneously.

Already in the works in the `new_sinks` branch but currently unusable (not finished at all, won't even compile).

> ASIO audio output

I don't see what the point of that would be? You can already use either rtaudio (default) or portaudio for output.

> RDS decoding should be more sensitive - better threshold for lower RF reception

The reason it appears "less sensitive" is the RDS specification is actually followed instead of letting any random rds blocks with errors through. You'll notice that sure, it's harder to receive but the quality of the data is much better with less or even no corrupted characters.

> dB signal power diagnostics

If you mean SNR, this is already shown on the SNR scale at the top and in the info text by hovering a VFO while holding the CTRL key. If you mean actual signal level in dBm like an S meter, this is just not possible with most SDRs. Companies like SDRplay get away with it in their software because their hardware is well tested and calibrated. Doing this in a general purpose software is futile at best and useless otherwise.

> SpyServer IP:PORT history (to be able to easily select between a few SpyServers

Could be considered but then it would have to be done on all networked sources for consistency, not just spyserver.

> Volume numbers

Not sure what you mean by that. If you mean number that indicate the volume "percent" it would honestly be kinda ugly and not really that useful. If you want to set it to some exact value you can always CTRL+click the slider and it'll let you enter a value between 0 and 1 but won't display it afterwards.

brane0690, 2023-05-24:
> > ASIO audio output

> I don't see what the point of that would be? You can already use either rtaudio (default) or portaudio for output.

It will enable lowest latency audio output for Windows users.

> > RDS decoding should be more sensitive - better threshold for lower RF reception

> The reason it appears "less sensitive" is the RDS specification is actually followed instead of letting any random rds blocks with errors through. You'll notice that sure, it's harder to receive but the quality of the data is much better with less or even no corrupted characters.

Sorry but I couldn't agree on this. On the same signal level, same station - SDR# decodes RDS without problems. On SDR++ no RDS at all.
Also some stations with higher RF signal (not the highest ones) do have random RDS blocks, compared to SDR# where RDS is error free.

> > dB signal power diagnostics

> If you mean SNR, this is already shown on the SNR scale at the top and in the info text by hovering a VFO while holding the CTRL key. If you mean actual signal level in dBm like an S meter, this is [...]

I mean on Signal Diagnostics like in SDR#. It is not enough to have real-time signal strenght dBFS readings for Filtered IF. There should be integration time about 10 seconds.

AlexandreRouma, 2023-05-24:
> It will enable lowest latency audio output for Windows users.

The audio output on windows already has no noticeable latency so I don't think this is worth implementing and maintaining.

> Sorry but I couldn't agree on this. On the same signal level, same station - SDR# decodes RDS without problems. On SDR++ no RDS at all.
Also some stations with higher RF signal (not the highest ones) do have random RDS blocks, compared to SDR# where RDS is error free.

I don't know where you're located but I'll assume in the US where a lot of stations don't understand that the RDS subcarrier is supposed to be in phase with the pilot just like the L-R subcarrier. Here in europe there is no such issue with the RDS.

> I mean on Signal Diagnostics like in SDR#. It is not enough to have real-time signal strenght dBFS readings for Filtered IF. There should be integration time about 10 seconds.

Well again, it's not measuring signal strength, it's measuring SNR right from the raw FFT of the baseband. Adding an integration time of 10 second would make it completely unusable for any kind of satellite aiming.

c3kkos, 2023-07-12:
Alexandre, ASIO support is useful for bypassing Windows mixer; doing so, the latency (on OS side, not on sdr++ side) would be lowered at its minimum (well, following ASIO's buffer setting)

So IT MIGHT be useful.

I used to be a wanna-be-musician and played a lot with DAWs like Reaper.

I KNOW exactly that in some cases you MUST deal with the lowest latency possible (eg. for live monitoring)
When you hit a guitar string you wanna listen at it in your monitor with no greater latency than -around- 10ms

For what concerns RF operations? 

My two cents: 
i don't care for extra latency if this is not going anywhere beyond 100ms. (which is already huge!)
If you deal with machines that recieve signals to switch something maybe it is critical.

To listen to airband or satcom? PMR LPD? No sense to talk about it.

THAT SAID:

A milestone in linux audio has been achieved recently.

It is called PIPEWIRE. A drop-in replacement for EVERYTHING that was in use till today.
A full replacement for pulseaudio and JACK. It deploys a layer upon ALSA and thats it.

Full node-control of sources and sink, for as many devices you want (audio, midi, video) with the lower latency possible.

I say THAT IS the future AND the way to go for HAM Radio
I encourage everyone to take it a shot in a modern linux distro.

themeadery, 2023-07-23:
Agree on the points about ASIO. If you are a musician or come from any type of DAW you want ASIO again in any other app you use because you get spoiled. It is superior in many ways including latency, which can definitely be important. Would love to see it in SDR++.

AlexandreRouma, 2023-07-23:
I won't support it, I'm not gonna be maintaining multiple versions of every audio related module for something almost nobody will notice anyway. If someone wants to make a third party plugin, have fun, not I'm not wasting time on it.