AlexandreRouma/SDRPlusPlus #1082: SDR++ Package doesn't use winehq-stable if it is installed

dscp46, 2023-05-29:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Intel Core i5-3320M @ 2.6GHz
- RAM: 8GB
- GPU: Intel HD Graphics 4000 (IVB GT2)
- SDR: Not applicable

**Software**
- Operating System: Debian GNU/Linux 12 (bookworm)
- SDR++: Official release 1.0.4 for Debian Sid (https://github.com/AlexandreRouma/SDRPlusPlus/releases/download/1.0.4/sdrpp_debian_sid_amd64.deb)

**Bug Description**
When installing sdrpp_debian_sid_amd64.deb, the package doesn't use winehq-stable as an alternative to wine-stable. WineHQ-stable is the official package, which is maintained by the developper team, and is required to get Winlink and VARA to work.

**Steps To Reproduce**
1. Install WineHQ-Stable, as explained on https://wiki.winehq.org/Debian
2. Download the last release of SDR++ from Github tagged versions.
3. sudo dpkg -i sdrpp_debian_sid_amd64.deb
4. sudo apt install -f (the request to uninstall winehq-stable to install wine-stable happens).

AlexandreRouma, 2023-05-29:
SDR++ doesn't use wine at all, I have no idea what you're talking about... Whatever your package manager is complaining about has nothing to do with SDR++.
Also, use the nightly, an old release like that doesn't work on sid since it's rolling release https://www.sdrpp.org/nightly.

closed 2023-05-29 by AlexandreRouma

OpenNetSurfer, 2024-08-18:
Hey @AlexandreRouma, this issue is referring to #718. I also faced the same problem.
I thought of simply mentioning here as no further comment was made. You also commented in that issue #718 a year ago from the time this issue was made, so I am just reminding that this is still happening in the latest build.

In short, it is a dependency problem with some audio library. 

A workaround for the problem is given [here](https://github.com/AlexandreRouma/SDRPlusPlus/issues/718#issuecomment-1115024671) in that same issue. This is a very crude workaround but it works (for me atleast).

Blueshawk, 2025-11-05:
"In short, it is a dependency problem with some audio library." - Yep, and it wrecks a wine setup too if you don't pay attention during install as Gdebi absolutely WILL remove things. (glares at doraemon)

This workaround also worked for install in Peppermint/Debian Bookworm with wine and winehq.

 Echoing here so more folks will find it more easily and maybe a dev will someday fix this in the dependency list.

```on May 2, 2022 · edited by RealJTG

Possible workaround (works for me in Ubuntu 20.04.4 with https://dl.winehq.org/wine-builds/ubuntu/ focal main repo in sources.list):

# This somehow provides libjack-dev without further conflicts :)
sudo apt-get install libjack-jackd2-dev

# librtaudio-dev seems to be happy now
sudo apt-get install librtaudio-dev

# Install sdrpp
sudo dpkg -i sdrpp_ubuntu_focal_amd64.deb

# Install missing dependencies
sudo apt --fix-broken install 

sdrpp
[2022-05-02 18:19:14.545] [info] SDR++ v1.0.4
...
[2022-05-02 18:19:15.758] [info] Ready.
^C

wine --version
wine-7.7

\o/
```
Link to original comment and context:
https://github.com/AlexandreRouma/SDRPlusPlus/issues/718#issuecomment-1115024671