AlexandreRouma/SDRPlusPlus #1083: 'core/src/dsp/chain.h' Strange wording

gvanem, 2023-05-31:
Not really a bug. But it seems strange that line 65 in [`core/src/dsp/chain.h`](https://github.com/AlexandreRouma/SDRPlusPlus/blob/master/core/src/dsp/chain.h#L65) says:
```c
        template<typename Func>
        void disableBlock(Processor<T, T>* block, Func onOutputChange) {
            // Check that the block is part of the chain
            if (!blockExists(block)) {
    -->         throw std::runtime_error("[chain] Tried to enable a block that isn't part of the chain");
            }
```

Should be `[chain] Tried to disable a block that isn't part of the chain` AFAICS (since the function is called `disableBlock()`).
Probably just an copy & paste issue from the function `enableBlock()` above.

gvanem, 2023-07-22:
*ping* anyone.

AlexandreRouma, 2024-01-22:
it's a typo not really a bug

closed 2024-01-22 by AlexandreRouma