AlexandreRouma/SDRPlusPlus #1084: Extra semicolon warnings

gvanem, 2023-05-31:
Another *not-really-a-bug report*:

But compiling SDR++ with `clang-cl -Wextra-semi-stmt  ..`, there is a large amount
of warnings like:
```
  warning: empty expression statement has no effect; remove unnecessary ';'
  to silence this warning [-Wextra-semi-stmt]
```

Only 2 of them are due to stb_image and the definition of the macro `dct_pass()`.
But `SET_DIFF_INT()` etc. could perhaps be defined as:
```c
  #define SET_DIFF_INT(id, in) do {                 \
          diffId = id;                              \
          diffValue.type = DRAW_LIST_ELEM_TYPE_INT; \
          diffValue.i = in;                         \
          syncRequired = elem.forceSync;            \
        } while (0)
```
All the unique and edited warnings I've found: 
<details>
<summary>extra-semi-stmt</summary>

```c
core/src/server.cpp(212,38):
            if (read < 0) { return; };
                                     ^
core/src/imgui/stb_image.h(2565,24):
   dct_pass(bias_0, 10);
                       ^
core/src/imgui/stb_image.h(2588,24):
   dct_pass(bias_1, 17);
                       ^
core/src/gui/smgui.cpp(80,67):
                    SET_DIFF_INT(elements[i].str, elements[i+1].i);
                                                                  ^
core/src/gui/smgui.cpp(86,51):
                    SET_DIFF_VOID(elements[i].str);
                                                  ^
core/src/gui/smgui.cpp(97,51):
                    SET_DIFF_VOID(elements[i].str);
                                                  ^
core/src/gui/smgui.cpp(109,67):
                    SET_DIFF_INT(elements[i].str, elements[i+1].i);
                                                                  ^
core/src/gui/smgui.cpp(115,69):
                    SET_DIFF_FLOAT(elements[i].str, elements[i+1].f);
                                                                    ^
core/src/gui/smgui.cpp(121,67):
                    SET_DIFF_INT(elements[i].str, elements[i+1].i);
                                                                  ^
core/src/gui/smgui.cpp(127,68):
                    SET_DIFF_BOOL(elements[i].str, elements[i+1].b);
                                                                   ^
core/src/gui/smgui.cpp(133,69):
                    SET_DIFF_FLOAT(elements[i].str, elements[i+1].f);
                                                                    ^
core/src/gui/smgui.cpp(142,61):
                    SET_DIFF_STRING(elements[i].str, tmpBuf);
                                                            ^
core/src/dsp/buffer/ring_buffer.h(137,48):
                if (toWrite < 0) { return -1; };
                                               ^
core/src/dsp/buffer/ring_buffer.h(175,48):
            if (lock) { _writable_mtx.lock(); };
                                               ^
core/src/dsp/buffer/ring_buffer.h(180,14):
                _readable_mtx.lock();
            };
             ^
core/src/dsp/buffer/ring_buffer.h(182,50):
            if (lock) { _readable_mtx.unlock(); };
                                                 ^
core/src/dsp/buffer/reshaper.h(123,80):
                if (ringBuf.readAndSkip(start, readCount, skip) < 0) { break; };
                                                                               ^
core/src/dsp/buffer/ring_buffer.h(72,47):
                if (toRead < 0) { return -1; };
                                              ^
core/src/dsp/buffer/ring_buffer.h(96,47):
                if (toRead < 0) { return -1; };
                                              ^
core/src/dsp/buffer/ring_buffer.h(125,48)
            if (lock) { _readable_mtx.lock(); };
                                               ^
core/src/dsp/buffer/ring_buffer.h(127,50):
            if (lock) { _readable_mtx.unlock(); };
                                                 ^
core/src/gui/widgets/waterfall.cpp(414,58):
                if (_fullUpdate) { updateWaterfallFb(); };
                                                         ^
core/src/gui/widgets/waterfall.cpp(442,58):
                if (_fullUpdate) { updateWaterfallFb(); };
                                                         ^
core/src/gui/widgets/waterfall.cpp(1018,50):
        if (_fullUpdate) { updateWaterfallFb(); };
                                                 ^
core/src/gui/widgets/waterfall.cpp(1040,50):
        if (_fullUpdate) { updateWaterfallFb(); };
                                                 ^
core/src/gui/widgets/waterfall.cpp(1077,50):
        if (_fullUpdate) { updateWaterfallFb(); };
                                                 ^
core/src/gui/widgets/waterfall.cpp(1090,50):
        if (_fullUpdate) { updateWaterfallFb(); };
                                                 ^
decoder_modules/radio/src/radio_module.h(195,10):
            _this->selectDemodByID(RADIO_DEMOD_CW);
        };
         ^
decoder_modules/radio/src/radio_module.h(202,10):
            _this->selectDemodByID(RADIO_DEMOD_RAW);
        };
         ^
source_modules/file_source/src/main.cpp(154,59):
            if (!_this->stream.swap(blockSize)) { break; };
                                                          ^
source_modules/file_source/src/main.cpp(168,59):
            if (!_this->stream.swap(blockSize)) { break; };
                                                          ^
decoder_modules/m17_decoder/src\m17dsp.h(494,76):
                lastConseqTime = std::chrono::high_resolution_clock::now();;
                                                                           ^
```
</details>

And some of these:
<details><summary>other warnings</summary>

```c
source_modules/spectran_http_source/src/spectran_http_client.cpp(53,88): warning: format specifies type 'int' but the argument has type
      'uint64_t' (aka 'unsigned long long') [-Wformat]
    sprintf(buf, "{\"frequencyCenter\":%d,\"frequencySpan\":%d,\"type\":\"capture\"}", freq, _samplerate);
                                       ~~                                              ^~~~
                                       %llu
source_modules/spectran_http_source/src/spectran_http_client.cpp(53,94): warning: format specifies type 'int' but the argument has type
      'uint64_t' (aka 'unsigned long long') [-Wformat]
    sprintf(buf, "{\"frequencyCenter\":%d,\"frequencySpan\":%d,\"type\":\"capture\"}", freq, _samplerate);
                                                            ~~                               ^~~~~~~~~~~
                                                            %llu
source_modules/spectran_http_source/src/spectran_http_client.cpp(56,27): warning: format specifies type 'int' but the argument has type
      'size_type' (aka 'unsigned long long') [-Wformat]
    sprintf(lenBuf, "%d", data.size());
                     ~~   ^~~~~~~~~~~
                     %zu

core/src\dsp/demod/ssb.h(116,9): warning: non-void function does not return a value in all control paths [-Wreturn-type]
        }
        ^
core/src/utils/proto/http.cpp(277,5): warning: non-void function does not return a value in all control paths [-Wreturn-type]
    }

decoder_modules/radio/src/demodulators/cw.h(21,37): warning: assigning field to itself [-Wself-assign-field]
            this->afbwChangeHandler = afbwChangeHandler;
                                    ^
```

</details>

Just a FYI.

AlexandreRouma, 2023-05-31:
I think all of those style warnings just show the reason pursuing an warningless clang compilation is futile. (I only look at gcc -Wall -Wextra warnings nowadays)

Both for stbimage and the smgui stuff, the styling makes perfect sense to a human reading the code, but to satisfy the compiler you'd have to hack garbage code to wrap the actual useful code, which although (normally) not impactful in terms of performance is confusing as hell to anyone attempting to understand the code.

For the other *actual* useless semicolons , I guess it was probably some typical 4am code

gvanem, 2023-06-01:
Style issue? If one would write things like:
``` c
if (x)
     dct_pass(bias_0, 10);
else dct_pass(bias_1, 10);
``` 
it wouldn't compile. But adding *hack garbage code* as you put it would work:
``` c
#define dct_pass(bias,shift) do { \
         ... \
        } while (0)
if (x)
     dct_pass(bias_0, 10);
else dct_pass(bias_1, 10);
``` 

So all these extra `;` seems sloppy. But not a big deal for me. Just thought you'd like to know.

closed 2024-01-22 by AlexandreRouma