AlexandreRouma/SDRPlusPlus #1086: SDR++ crashes when I switch back from background

bczhc, 2023-06-01:
**Hardware**
- CPU: 11th Gen Intel i7-11800H (16) @ 3.000GHz
- RAM: 64GB
- GPU: Intel TigerLake-H GT1 [UHD Graphics]
- SDR: Local; both RTL-SDR and RSP1A

**Software**
- Operating System: Arch Linux rolling, kernel 6.3.4; WM: i3, without compositor
- SDR++: v1.1.0, built at May 31 2023

**Bug Description**
See below.

**Steps To Reproduce**
1. Open SDR++ and tune to some FM and listen it
2. Put it at the background an do other things
3. After a while then switch back, finding it immediately hangs and the FM sound stops

**Additional info**
I don't know how long precisely I put it at the background for, then it will crash. I just encounter this issue from time to time.

Coredump from `coredumpctl`
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Core was generated by `sdrpp'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f9bc010bca3 in XSetICFocus (ic=0x56297eabc490) at xlibi18n/ICWrap.c:349
Downloading source file /usr/src/debug/libx11/libX11-1.8.4/src/xlibi18n/ICWrap.c
349	      (*ic->methods->set_focus) (ic);
[Current thread is 1 (Thread 0x7f9bc1461740 (LWP 924291))]
(gdb)
(gdb) bt
#0  0x00007f9bc010bca3 in XSetICFocus (ic=0x56297eabc490) at xlibi18n/ICWrap.c:349
#1  XSetICFocus (ic=0x56297eabc490) at xlibi18n/ICWrap.c:346
#2  0x00007f9bc14be647 in ?? () from /usr/lib/libglfw.so.3
#3  0x00007f9bc0ff52fe in backend::renderLoop() () from /usr/lib/libsdrpp_core.so
#4  0x00007f9bc0e8f41e in sdrpp_main(int, char**) () from /usr/lib/libsdrpp_core.so
#5  0x00007f9bc0c39850 in __libc_start_call_main (main=main@entry=0x56297cf73020, argc=argc@entry=1, argv=argv@entry=0x7ffd1d90daf8)
    at ../sysdeps/nptl/libc_start_call_main.h:58
#6  0x00007f9bc0c3990a in __libc_start_main_impl (main=0x56297cf73020, argc=1, argv=0x7ffd1d90daf8, init=<optimized out>,
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffd1d90dae8) at ../csu/libc-start.c:360
#7  0x000056297cf73055 in ?? ()
(gdb) l
344	 */
345	void
346	XSetICFocus(XIC ic)
347	{
348	  if (ic && ic->core.im)
349	      (*ic->methods->set_focus) (ic);
350	}
351	
352	/*
353	 * Require the input manager to unfocus the focus window attached to the ic
(gdb)
```

bczhc, 2023-06-03:
Caught it. It crashes when I switch from workspace 10 to 1.

https://github.com/AlexandreRouma/SDRPlusPlus/assets/49330580/9e48da0f-e546-4ff6-828a-edfb851b8bbf

AlexandreRouma, 2023-06-03:
Can't replicate at all (also on arch and i3). Backtrace says it's in X11 so probably more an issue on that side. I'll wait for other people to try to repro but almost certainly not an issue with SDR++ so I'll close if nobody else can replicate or if it's proven to be in xorg.

bczhc, 2023-06-21:
Weird. I just didn't encounter this issue these days not knowing why and from when. As you said I just close this issue now :).

closed 2023-06-21 by bczhc

bczhc, 2023-07-22:
I'm here again, and I found the issue cause! This can have a stable reproducibility. The step is:
1. start [Fcitx5](https://github.com/fcitx/fcitx5)
1. start SDR++
1. move the focus out of SDR++ and stop Fcitx5
1. focus back on SDR++, and it will crash

(If I swap step 1 and 2, this issue won't happen.)

A screenrecord demonstration (using a demo from ImGui):

[a.webm](https://github.com/AlexandreRouma/SDRPlusPlus/assets/49330580/7889a7dc-894a-4489-96c5-c15d1c8fe18d)

This is from the upstream GLFW windowing stuff. After bisecting, the fixed commit is https://github.com/glfw/glfw/commit/44b5d06583cd21ac237eb8f6263db03faf1726c2. But unfortunately, this fix hasn't been included in the latest 3.3.8 GLFW version (patch branch for 3.3).

For me to fix this issue, just `yay -S glfw-git` :).