AlexandreRouma/SDRPlusPlus #1087: SDRplay frequency offset

bczhc, 2023-06-02:
**Hardware**
- CPU: 11th Gen Intel i7-11800H (16) @ 3.000GHz
- RAM: 64GB
- GPU: Intel TigerLake-H GT1 [UHD Graphics]
- SDR: Local; RSP1A

**Software**
- Operating System: Arch Linux rolling, kernel 6.3.4; WM: i3, without compositor
- SDR++: v1.1.0, built at May 31 2023

**Bug Description**
When I rapidly drag the frequency bar, it has some chance that the frequency will have offset. This reproduces randomly. I tried so many times to trigger this issue, then cut the part out, as the screenrecord shown below.

**Additional info**
Screenrecord:

https://github.com/AlexandreRouma/SDRPlusPlus/assets/49330580/48107f9d-99f1-4047-8baf-48fae01d79c7

You can see this FM radio is at 105.2MHz in the beginning. After doing some messy movement, that radio lies at 105.7MHz, where this issue happens. When I stop and then restart the SDR, it turns back OK.

starvald, 2023-06-02:
i have seen this a lot too, i think enabling 'IF AGC' makes it more likely to happen.

AlexandreRouma, 2023-06-02:
Can only be a bug in the SDRplay API. All tuning in SDR++ is done synchronously so this kind of thing cannot happen.

They're probably doing USB communications in a non-thread-safe manner and end up with data races when the center frequency is changing rapidly.

I recommend you try replicating this issue on another software that allows to change the center frequency quickly like that.
When you're able to show the same issue, report it to SDRplay (otherwise they'll just dismiss it saying it's SDR++)

bczhc, 2023-06-02:
Yea, I just did reproduce the issue with CubicSDR. Then I'll ask for help at SDRplay side and close this issue.

closed 2023-06-02 by bczhc