AlexandreRouma/SDRPlusPlus #1088: too small window causes crash

bczhc, 2023-06-03:
**Hardware**
- CPU: 11th Gen Intel i7-11800H (16) @ 3.000GHz
- RAM: 64GB
- GPU: Intel TigerLake-H GT1 [UHD Graphics]

**Software**
- Operating System: Arch Linux rolling, kernel 6.3.4; WM: i3, without compositor
- SDR++: v1.1.0, built at June 2 2023

**Bug Description**
See below.

**Steps To Reproduce**
1. Start SDR++
2. Resize window
3. Crashes when the window size is too small

**Additional info**
This is 100% reproducible.

Coredump:
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Core was generated by `sdrpp'.
Program terminated with signal SIGABRT, Aborted.
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44	      return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;
[Current thread is 1 (Thread 0x7ff45d29d740 (LWP 916283))]
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ff45ca9f2d3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ff45ca4fa08 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ff45ca38538 in __GI_abort () at abort.c:79
#4  0x00007ff45bcdd3b2 in std::__glibcxx_assert_fail (file=<optimized out>, line=<optimized out>, function=<optimized out>,
    condition=<optimized out>) at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/debug.cc:61
#5  0x00007ff45cd0fda0 in ImGui::WaterFall::drawFFT() () from /usr/lib/libsdrpp_core.so
#6  0x00007ff45cd1638e in ImGui::WaterFall::draw() () from /usr/lib/libsdrpp_core.so
#7  0x00007ff45ccc1786 in MainWindow::draw() () from /usr/lib/libsdrpp_core.so
#8  0x00007ff45cdf5598 in backend::renderLoop() () from /usr/lib/libsdrpp_core.so
#9  0x00007ff45cc8f41e in sdrpp_main(int, char**) () from /usr/lib/libsdrpp_core.so
#10 0x00007ff45ca39850 in __libc_start_call_main (main=main@entry=0x55f81994d020, argc=argc@entry=1, argv=argv@entry=0x7fff1d780d98)
    at ../sysdeps/nptl/libc_start_call_main.h:58
#11 0x00007ff45ca3990a in __libc_start_main_impl (main=0x55f81994d020, argc=1, argv=0x7fff1d780d98, init=<optimized out>,
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fff1d780d88) at ../csu/libc-start.c:360
#12 0x000055f81994d055 in ?? ()
(gdb) l
39	         delivery of all pending signals after unblocking in the code
40	         below.  POSIX only guarantees delivery of a single signal,
41	         which may not be the right one.)  */
42	      pid_t tid = INTERNAL_SYSCALL_CALL (gettid);
43	      int ret = INTERNAL_SYSCALL_CALL (tgkill, __getpid (), tid, signo);
44	      return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;
45	    }
46	
47	  /* Block all signals, as required by pd->exit_lock.  */
48	  internal_sigset_t old_mask;
(gdb) quit
```

Screenrecord:

https://github.com/AlexandreRouma/SDRPlusPlus/assets/49330580/98612738-ebdc-4720-8d1d-c865776ec494

bczhc, 2023-06-03:
This may be related to #689 ?

AlexandreRouma, 2023-06-03:
known issue, won't fix since it'll just be fixed with the new waterfall and is an issue to absolutely nobody. Just don't run at silly window sizes.

closed 2023-06-03 by AlexandreRouma