AlexandreRouma/SDRPlusPlus #1090: HackRF (with and without PP) on Android

sizzlik, 2023-06-04:
Hey,
I got 2 hRF One (r9) boards, one with portapack H2+, one just in alu enclosure. Both are on the newest firmware and both work fine on my windows pc. But on Android the app just closes when i click play (on phone and tablet). Works with the RTL-SDR V3, nooelec smart xtr and smart v5.
HackRF works with other apps like sdrAngel, SDRtouch etc.
Same on DragonOS Focal, but that version might be outdated, got to check.

AlexandreRouma, 2023-06-04:
Both my hackrfs work fine for me on android (latest firmware). can't replicate at all.

sizzlik, 2023-06-04:
tried again..works on windows and dragonOS. But for some reason not on my Samsung Tab A 2019 nor my Xiaomi Note 8T. All other SDR that i got (RTL V3, Nooelec V5, Nooelec XTR and a noname china bugger work..only the hackRF closes the app once i click play. But it recognizes the hRF when i click refresh. Only crashes when i click play.
Sure the problem must be on my side..any idea what it could be maybe? As other app like SDRAngel, touchSDR, MagicSDR work on android with the hRF.

AlexandreRouma, 2023-06-05:
if you have some other app that takes control of the hackrf by force (by listing it in its app manifest) it could cause issues. Try it on a device with no other SDR app installed.

Dane-Law, 2023-06-23:
Do you have RF Analyzer installed on your Android devices?
this fellow ?? https://i.imgur.com/RMGEJWS

sizzlik, 2023-06-23:
Nope..only dump1090 and a ads-b app, wich i uninstalled to test but no change. I guess its a problem with my hRF board.

Dane-Law, 2023-06-23:
I got the exact same problem, I have tried it on 3 different Android devices.
* Samsung Tab S5e tablet 10.5" Tablet
* Xiaoxin P11pro2021 11.5" Tablet
* Samsung Galaxy S9+ 6.2" mobile (completely clean, no SDR tools have ever been installed)

and it's the same on all, it will install and everything fine, and also ask if I want HackRF to connect, and it will process with hackRF USB when refresh, but the moment you then click the play button, it hard closes the SDR++ app.

So I reckon its something with the HackRF, maybe "HackRF in Portapack with mayhem fw simply dont work, am on mayhem fw 1.2-.
weird as Hack with Portapack it works fine in Windows with SDR++ windows version and other HackRF android apps like RF analyzer, that I had installed on the S5e tablet.. also works great.

This is my HackRF Portapack version, which is the most common and most popular recommended version from the HackRF Portapack community (Model HackRF Portapack H2, and I got Mayhem 1.2 fw on it, and eveverything else works great, only SDR++ I cant get to work in Android as it hard closes the app, in windows no problem.)
https://i.imgur.com/9H7j7Yk.jpg
https://i.imgur.com/CRwdrJ5.jpg
https://i.imgur.com/jzNXNP6

Any approach to fix this?

Dane-Law, 2023-06-27:
> Both my hackrfs work fine for me on android (latest firmware). can't replicate at all.

what HackRF do you have?
and is it standalone original units?

AlexandreRouma, 2023-06-27:
both a clone and an official one.

sizzlik, 2023-06-27:
i got 2 clones, both R9, different manufactur, same chipset (MAX2839/Si5351A/WM8731/RFFC5072), one in a portapack H2+, one just in an alu encolure. The "original" is not different nowadays due to the recent chip shortage, and they all come from the same factory in shenzeng. Afaik the R9 started to use the chips that others used previously.

derkommutator, 2023-06-28:
I'm experiencing exactly the same behaviour with a recently-acquired HackRF (original I think, 2018.01.1 firmware) on both a Lenovo P11 Pro running factory Android 11, as well as a Fxtec Pro1-X running LineageOS 20.

I found a copy of the old "original" March 2022 prerelease of SDR++ for Android, and it does _not_ exhibit this problem. Everything appears to be working properly in that build.

sizzlik, 2023-06-28:
> I'm experiencing exactly the same behaviour with a recently-acquired HackRF (original I think, 2018.01.1 firmware) on both a Lenovo P11 Pro running factory Android 11, as well as a Fxtec Pro1-X [...]

Mind to upload the apk somewhere, so i can test aswell? Maybe there is a chance we find the problem when comparing the changes in the build.

derkommutator, 2023-06-28:
> Mind to upload the apk somewhere, so i can test aswell? Maybe there is a chance we find the problem when comparing the changes in the build.

There is a Google Drive link on this page:  https://forums.qrz.com/index.php?threads/sdr-plus-plus-for-android-lovers.806694/

Naturally I make no claims as to the purity of the software behind that link. Probably fine, but this should not be considered a trustworthy source. ;)

sizzlik, 2023-06-28:
> There is a Google Drive link on this page: https://forums.qrz.com/index.php?threads/sdr-plus-plus-for-android-lovers.806694/
>
> Naturally I make no claims as to the purity of the software behind that [...]

VM >> BlueStacks >> wireshark >> see if E.T. wants to phone home

derkommutator, 2023-06-28:
> VM >> BlueStacks >> wireshark >> see if E.T. wants to phone home

Well, my device is rooted and firewalled, and doesn't have anything personal on it anyway. Really just issuing a standard "caveat emptor". :)

sizzlik, 2023-06-28:
Hehe..i ment that as general advice for people that want to be extra secure if they browse this page ;)
Nothing to gain from my devices either, but better save than sorry.

Dane-Law, 2023-06-28:
> There is a Google Drive link on this page: https://forums.qrz.com/index.php?threads/sdr-plus-plus-for-android-lovers.806694/
>
> Naturally I make no claims as to the purity of the software behind that [...]

That worked, thx buddy.
Tried it with my HackRF Portapack (model pictured above) on a XiaoxinP11pro2021 (aka LenovoP11pro2021)
now the SDR++ android app, dont crash the moment, I push the play button after initialised the (HackRF USB)
Dont have the skill & insight level to decipher what exactly the dev have changed since this android-prerelease that makes current android versions crash and the old 2022 apk dont..

sizzlik, 2023-06-28:
didnt had time to test yet, but we got 3 people with the same issue and apparently the issue isnt the hardware but some change that happened some time ago. Maybe Alexandre still got the source from that or another past version to compare..or maybe even remembers what changes had been made in between. Last resort, disassemble, bit comparison, lots of coffee and a few prayers. But i think we getting closer to a bugfix.

Do you guys got any problems with the pc versions? On dragonOS i get the same..it worked in the past, doesnt work anymore. Dont have windows to test 
atm, but can set up one of my laptops with a fresh win if needed.

derkommutator, 2023-06-28:
> Maybe Alexandre still got the source from that or another past version to compare..or maybe even remembers what changes had been made in between. Last resort, disassemble, bit comparison, lots of coffee and a few prayers.

Hahah that's a little extreme. All the code commits are right here on github, so it's certainly possible to narrow down where it broke. However, a scattershot approach to the problem is likely not the fastest way to get to a solution. Probably best at this point for someone familiar with the code evolution to inject some thoughts.

sizzlik, 2023-06-29:
Maybe I should end my comments with /S to indicate when im jokin. Youre right, Alex needs to have a look as he knows best. But its good, thanks to you, that we now have an anchor point to look at.

sizzlik, 2023-07-05:
Dont know if its of any help, but here is a terminal output what happens on my side, runs fine until it closes with the final line "Segmentation Fault". There are a few errors while loading up aswell. Taken from a fresh updated DragonOS install.

moe@radioshack:~$ sudo sdrpp
[05/07/2023 06:48:27.000] [INFO] SDR++ v1.1.0
[05/07/2023 06:48:27.000] [INFO] Loading config
[05/07/2023 06:48:28.000] [INFO] Using OpenGL 3.0
[05/07/2023 06:48:28.000] [INFO] Loading icons
[05/07/2023 06:48:28.000] [INFO] Loading band plans
[05/07/2023 06:48:28.000] [INFO] Loading band plans color table
[05/07/2023 06:48:28.000] [ERROR] Menu element is missing name key
[05/07/2023 06:48:28.000] [INFO] Loading modules
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_client.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/radio.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/m17_decoder.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/recorder.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[05/07/2023 06:48:28.000] [WARN] ConfigManager locked, waiting...
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/bladerf_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_sink.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/scanner.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/inmarsatc_demodulator.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/file_source.so
[05/07/2023 06:48:28.000] [INFO] Loading /usr/lib/sdrpp/plugins/perseus_source.so
[05/07/2023 06:48:28.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/perseus_source.so: libperseus-sdr.so.0: cannot open shared object file: No such file or directory
[05/07/2023 06:48:28.000] [INFO] Initializing Airspy Source (airspy_source)
[05/07/2023 06:48:28.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[05/07/2023 06:48:28.000] [INFO] Initializing Audio Sink (audio_sink)
[05/07/2023 06:48:28.000] [INFO] Initializing Audio Source (audio_source)

RtApiPulse::DeviceInfo pa_context_connect() failed: Connection refused

[05/07/2023 06:48:28.000] [INFO] New DSP samplerate: 48000.000000 (source samplerate is 48000.000000)
[05/07/2023 06:48:28.000] [INFO] Initializing BladeRF Source (bladerf_source)
[05/07/2023 06:48:28.000] [ERROR] Could not list devices -7
[05/07/2023 06:48:28.000] [INFO] Initializing File Source (file_source)
[05/07/2023 06:48:28.000] [INFO] Initializing Frequency Manager (frequency_manager)
[05/07/2023 06:48:28.000] [INFO] Initializing HackRF Source (hackrf_source)
[05/07/2023 06:48:28.000] [INFO] Initializing Hermes Source (hermes_source)
[05/07/2023 06:48:28.000] [INFO] Initializing LimeSDR Source (limesdr_source)
[05/07/2023 06:48:28.000] [INFO] Initializing Network Sink (network_sink)
[05/07/2023 06:48:28.000] [INFO] Initializing PerseusSDR Source (perseus_source)
[05/07/2023 06:48:28.000] [ERROR] Module 'perseus_source' doesn't exist
[05/07/2023 06:48:28.000] [INFO] Initializing PlutoSDR Source (plutosdr_source)
[05/07/2023 06:48:28.000] [INFO] Initializing RFspace Source (rfspace_source)
[05/07/2023 06:48:28.000] [INFO] Initializing RTL-SDR Source (rtl_sdr_source)
[05/07/2023 06:48:28.000] [INFO] Initializing RTL-TCP Source (rtl_tcp_source)
[05/07/2023 06:48:28.000] [INFO] Initializing Radio (radio)
[Resamp] predec: 8192, interp: 25, decim: 6, inacc: 0.000000%, taps: 1900
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500

[... 42 lines cut ...]
[05/07/2023 06:48:30.000] [INFO] Running post-init for Airspy Source
[05/07/2023 06:48:30.000] [INFO] Running post-init for AirspyHF+ Source
[05/07/2023 06:48:30.000] [INFO] Running post-init for Audio Sink
[... cut ...]
[05/07/2023 06:48:30.000] [INFO] Running post-init for Radio
[05/07/2023 06:48:30.000] [INFO] Running post-init for Recorder
[05/07/2023 06:48:30.000] [INFO] Running post-init for Rigctl Server
[05/07/2023 06:48:30.000] [INFO] Running post-init for SDR++ Server Source
[05/07/2023 06:48:30.000] [INFO] Running post-init for SDRplay Source
[05/07/2023 06:48:30.000] [INFO] Running post-init for SoapySDR Source
[05/07/2023 06:48:30.000] [INFO] Running post-init for SpyServer Source
[05/07/2023 06:48:30.000] [INFO] Ready.
Segmentation fault
moe@radioshack:~$

Dane-Law, 2023-11-02:
Any dev-conclusion why, so many HackRF H2+ (portapack Mayhem OS) had problems in connecting with newer "SDR++ builds", and the very old prerelease version worked fine?

AlexandreRouma, 2023-11-02:
The hackrf code hasn't changed in years. Something changed on your end, likely firmware version.

Dane-Law, 2023-11-02:
As you can see it's just not me, it's all the people in this thread, that got the exact same issue and couldnt connect and some people got more than 1 HackRF just like you Alexandre.
The moment you tried to connect and run it, the software would hard close and it happen on multiple different tablets from Lenovo/Xiaoxin to Samsung etc.
People then found the prerelease version from years back, and that weirdly enough worked fine, so people are just using that, but it's an old version that likely lacks some of the things the newer version got.

I still have the prerelease version on my tablet to use with my HackRF (you can see all the board pictures and details above) and was simply wondering if you the dev had concluded what differed from the prerelease version to the version that we all had problem with, and would hardclose over and over, when connecting , and if the latest SDR++ perhaps had that part sorted.

Dane-Law, 2023-11-02:
But will try to install the latest SDR++ apk on another tablet and give it a fresh shot from the bow but the very old prerelease verso. worked fine 
https://i.imgur.com/8EYK9BA.

AlexandreRouma, 2024-01-29:
Have had no adidtional reports since then and never been able to replicated on any device using any of my hackrfs.
I have no choice but to close this and wait to see if someone else reports an issue similar.

closed 2024-01-29 by AlexandreRouma