AlexandreRouma/SDRPlusPlus #1092: Rig Failure w/ WSJTX v2.6.1

alphafox02, 2023-06-06:
**Hardware**
- CPU: i7 
- RAM: 8GB
- GPU: Nvidia
- SDR: Local

**Software**
- Operating System: Ubuntu 22.04
- SDR++: v1.1.0 Nightly built May 17

**Bug Description**
Changing from WSJTX 2.5.4 to 2.6.1 results in the inability for WSJTX to interact over Hamlib NET rigctl with SDRPP. Previous 2.5.4 had no issues. Also checked and found that GQRX is able to connect with 2.6.1 WSJTX, at least as far as I can tell as changes within WSJTX reflect in GQRX. 

**Steps To Reproduce**
1. Install latest Jammy SDRPP Nightly
2. Install WSJTX 2.6.1 deb package from WSJTX site
3. Start SDRPP rig ctrl server and attempt to TEST CAT from WSJTX to the IP/port of SDRPP.

**Screenshots**
[image]

**Additional info**
SDRPP output in console from the point of starting and attempting to connect to rigctl

[06/06/2023 06:17:21.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Tune: 100000000.000000!
[06/06/2023 06:17:29.000] [INFO] New DSP samplerate: 2000000.000000 (source samplerate is 2000000.000000)
[06/06/2023 06:17:33.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Start!
[06/06/2023 06:17:33.000] [INFO] SDRPlaySourceModule 'SDRplay Source': Tune: 100000000.000000!
[06/06/2023 06:17:51.000] [INFO] Rigctl command: '\chk_vfo'
[06/06/2023 06:17:51.000] [INFO] Rigctl command: '\dump_state'
[06/06/2023 06:17:51.000] [INFO] Rigctl command: 'v'
[06/06/2023 06:17:51.000] [INFO] Rigctl command: 'f VFOA'
[06/06/2023 06:17:51.000] [INFO] Rigctl command: 'f VFOB'
[06/06/2023 06:17:51.000] [INFO] Rigctl command: 's VFOA'
[06/06/2023 06:17:51.000] [INFO] Rigctl command: 'V VFOA'
[06/06/2023 06:17:51.000] [INFO] Rigctl command: 'm VFOA'
[06/06/2023 06:17:51.000] [INFO] Rigctl command: 'F VFOA 100031255.000000'
[06/06/2023 06:17:51.000] [INFO] Rigctl command: 'q'
[06/06/2023 06:18:01.000] [INFO] Rigctl command: '\chk_vfo'
[06/06/2023 06:18:01.000] [INFO] Rigctl command: '\dump_state'
[06/06/2023 06:18:01.000] [INFO] Rigctl command: 'v'
[06/06/2023 06:18:01.000] [INFO] Rigctl command: 'f VFOA'
[06/06/2023 06:18:01.000] [INFO] Rigctl command: 'f VFOB'
[06/06/2023 06:18:01.000] [INFO] Rigctl command: 's VFOA'
[06/06/2023 06:18:01.000] [INFO] Rigctl command: 'V VFOA'
[06/06/2023 06:18:01.000] [INFO] Rigctl command: 'm VFOA'
[06/06/2023 06:18:01.000] [INFO] Rigctl command: 'F VFOA 100031255.000000'
[06/06/2023 06:18:01.000] [INFO] Rigctl command: 'q'
[06/06/2023 06:18:11.000] [INFO] Rigctl command: '\chk_vfo'
[06/06/2023 06:18:11.000] [INFO] Rigctl command: '\dump_state'
[06/06/2023 06:18:11.000] [INFO] Rigctl command: 'v'
[06/06/2023 06:18:11.000] [INFO] Rigctl command: 'f VFOA'
[06/06/2023 06:18:11.000] [INFO] Rigctl command: 'f VFOB'
[06/06/2023 06:18:11.000] [INFO] Rigctl command: 's VFOA'
[06/06/2023 06:18:11.000] [INFO] Rigctl command: 'V VFOA'
[06/06/2023 06:18:11.000] [INFO] Rigctl command: 'm VFOA'
[06/06/2023 06:18:11.000] [INFO] Rigctl command: 'F VFOA 100031255.000000'
[06/06/2023 06:18:11.000] [INFO] Rigctl command: 'q'

alphafox02, 2023-06-06:
Only differences in WSJTX that seem to be related to Rig CTL is

 - Fix a hamlib bug caused by setting VFO B when Split is not used
 - Rig control interface to Ham Radio deluxe made more robust 
 - Hamlib bug fixes: make sure that VFOB is selected when required on
   some rigs (e.g. IC7610, IC7100), and others.

alphafox02, 2023-06-07:
As a follow up, this issue goes away after installing and testing SDR++ w/ WSJT-X 2.7.0-rc1. So maybe this is more of a bug in WSJTX?

gprime31, 2023-07-21:
Also:
Working with wsjtx 2.5.2
Not working with JTDX 2.2.159

themeadery, 2023-07-23:
> As a follow up, this issue goes away after installing and testing SDR++ w/ WSJT-X 2.7.0-rc1. So maybe this is more of a bug in WSJTX?

Upgraded WSJT-X from 2.6.1 stable to 2.7.0-rc2 and all of a sudden rigctl works! It still had some errors, but I slowed down and let it sit for a while, made deliberate changes, and now it's working properly.

I was digging into this bug a couple days ago and it all seems related to the built-in copy of hamlib that WSJT-X was including in their install. It was old or something and had bugs. It seems like the RC's of the new version have newer hamlib dll's that work better with SDR++ now. I think maybe SDR++ updated and WSJT-X was lagging behind on the hamlib library.

gprime31, 2023-07-23:
Ya same with JTDX

https://wsjtx.groups.io/g/main/topic/just_did_update_to_latest/91570250?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,0,91570250,previd%3D9223372036854775807,nextid%3D1654291319721204427&previd=9223372036854775807&nextid=1654291319721204427

gprime31, 2023-07-23:
Can download the libham for win from ^^ 
Linux just rebuild from source. 
When WSJT-X and JTDX release the next version won't have to mess with anything.

themeadery, 2023-07-23:
I was too lazy and scared to just start copying random hamlib DLL's into the WSJT-X directory, but glad to know that works, too.

AlexandreRouma, 2024-01-29:
since a new hamlib or WSJT-X fixes the issue, I think this issue can be closed.

closed 2024-01-29 by AlexandreRouma