AlexandreRouma/SDRPlusPlus #1093: DSP dependencies broken for modules: Weather Sat, Falcon9, KG SSTV

Szewcson, 2023-06-09:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: 
- RAM:
- GPU: 
- SDR: (Remote or local? If remote, what protocol?)

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

**Bug Description**
DSP sources dir structure and API was changed but this changes are not implemented in decoder modules. They can't be build at the moment, and changes are to big to quickfix it without any knowladge about the code base.

**Steps To Reproduce**
1. execute `cmake .. -DCMAKE_INSTALL_PREFIX=/usr/local -DOPT_BUILD_METEOR_DEMODULATOR=ON -DOPT_BUILD_KG_SSTV_DECODER=ON -DOPT_BUILD_WEATHER_SAT_DECODER=ON -DOPT_BUILD_FALCON9_DECODER=ON -DOPT_BUILD_ATV_DECODER=ON -DOPT_BUILD_M17_DECODER=ON`
2. execute `make`
3. see the errors about missing dsp headers

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

AlexandreRouma, 2023-06-11:
A lot of the modules you tried to build are unfinished and therefor build should not be attempted. A lot of these (falcon9_decoder, weather_sat_decoder and kg_sstv_decoder) will also be removed and archived soon.

See https://github.com/AlexandreRouma/SDRPlusPlus#module-list for details on the state of development of various modules.

closed 2023-06-11 by AlexandreRouma

Szewcson, 2023-06-13:
THX for the replay. In fact I was only hoping to use weather_sat_decoder (the others I tried to build out of curiosity). Sad to hear that this module will be removed.