AlexandreRouma/SDRPlusPlus #1097: SDR++ Android not starting when pressing the "play" button

RFFreebander433, 2023-06-17:
Hello, I installed the SDR++ apk and it downloaded fine, I've selected the source (RTL-SDR), set my bandwidth to 2.048 and selected direct sampling, aswell set the frequency to a time signal (9.996 MHz RWM Time signal), I've set all that and pressed the play button, the spectrum does not start and there is no audio even.

OS: Android 13

The SDR is plugged in fine, even the app detects it but it doesn't start up when pressing the play button. Am I missing something here or is this a bug?

AlexandreRouma, 2023-06-17:
Show a screenshot that includes the waterfall and source menu.

RFFreebander433, 2023-06-18:
[image]

AlexandreRouma, 2023-06-18:
It's not detected otherwise it would show up in the dropdown. "RTL-SDR" is the source type, not the device.
Probably forgot to press refresh which is required on android after giving device permissions.
See the manual for details https://www.sdrpp.org/manual.pdf

RFFreebander433, 2023-06-18:
I tried refresh and it worked! Thanks for the help!

RFFreebander433, 2023-06-18:
Also, this SDR++ android is a really a great way to bring my SDR outdoors for better reception, thank you for making this!

closed 2023-06-18 by AlexandreRouma