AlexandreRouma/SDRPlusPlus #1099: App starts with microscopic font size

flybd5, 2023-06-19:
Why does this application start up with a microscopic, unreadable font for the UI? No, my graphics drivers do not need updating, and the "Display' menu shows nothing about font size. The app also does not respond to the standard Ctrl-plus and Ctrl-minus to manage font sizes. ???

**Hardware**
- CPU: Xeon 32 cores
- RAM: 64 gb
- GPU:  NVidia 2060
- SDR: not relevant, it's not configured to use it because I can't read the text

**Software**
- Operating System: Win 10 latest
- SDR++: Latest build as of today

AlexandreRouma, 2023-06-19:
>  standard Ctrl-plus and Ctrl-minus to manage font sizes.

There is no such standard and switching font size requires a restart anyway.

The manual's troubleshooting section clearly talks about this: https://www.sdrpp.org/manual.pdf page 71

SDR++ Display menu -> High-DPI scaling -> set it to 200 or 300% and restart SDR++.

closed 2023-06-19 by AlexandreRouma

flybd5, 2023-06-19:
Switched to better software that doesn't require jumping through hoops or restarting to manage font size, and easily detects my SDR. See ya.