AlexandreRouma/SDRPlusPlus #1110: MacOS Ventura: SDR++ fails to run

RCCRAFT1, 2023-06-30:
Was attempting to get SDR++ to work on my MacBook M2 with Ventura installed and I get this error when running it from the terminal

mrc402@MacBook-Air ~ % sdrpp
dyld[53999]: Library not loaded: /usr/local/opt/glew/lib/libGLEW.2.2.dylib
  Referenced from: /usr/local/bin/sdrpp
  Reason: tried: '/usr/local/opt/glew/lib/libGLEW.2.2.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/glew/lib/libGLEW.2.2.dylib' (no such file), '/usr/local/opt/glew/lib/libGLEW.2.2.dylib' (no such file), '/usr/local/lib/libGLEW.2.2.dylib' (no such file), '/usr/lib/libGLEW.2.2.dylib' (no such file, not in dyld cache)
zsh: abort      sdrpp

I used the MacOS PKG file that was included in the latest release.

AlexandreRouma, 2023-06-30:
Use the nightly, not the pkg release: https://www.sdrpp.org/nightly

closed 2023-06-30 by AlexandreRouma

RCCRAFT1, 2023-06-30:
You sure that supports Apple silicon?

AlexandreRouma, 2023-07-01:
the PKG doesn't support it either.

RCCRAFT1, 2023-07-01:
Then if you aren't supposed to use the release version of the program why does a Mac pkg exist under that release name?

AlexandreRouma, 2023-07-01:
Because the MacOS packaging was fixed after that release was made...?

RCCRAFT1, 2023-07-01:
Then just use the nightly and don't mark it as a pre-release lol

AlexandreRouma, 2023-07-01:
Guess what the readme and manual recommends...
It's not yet ready feature wise for me to create a new release.