AlexandreRouma/SDRPlusPlus #1111: Configurable smoothing for SNR meter

thehans, 2023-07-01:
**Feature description**
The SNR meter jumps around so quickly, it makes it difficult to use as a measure of signal quality when tweaking things.
It would be nice if this display could be smoothed over time (moving average) by some configurable amount, much like the waterfall and FFT are done.

Also a digital readout of SNR, in addition to the bar meter, would be nice for making more quantitative readings, to facilitate before/after type of comparisons.

AlexandreRouma, 2023-07-07:
Could be considered but there are many ways to average which are useful in different situations (peak and average). I'll wait to see if anyone else requests this.

> Also a digital readout of SNR, in addition to the bar meter, would be nice for making more quantitative readings, to facilitate before/after type of comparisons.

Press CTRL while hovering the VFO with your mouse.

AlexandreRouma, 2023-07-10:
Impemented

closed 2023-07-10 by AlexandreRouma