AlexandreRouma/SDRPlusPlus #1116: Fractional UI scaling support

DuMOHsmol, 2023-07-02:
At the moment, SDR++ allows only integer UI scaling options: 100%, 200%, etc. But for some displays, 100% is too small and is too 200% huge. It would be nice to gain some level of fractional scaling support.

DuMOHsmol, 2023-07-02:
Currently, the fractional scaling options can't be added right away because some widgets become misaligned (see discussion in https://github.com/AlexandreRouma/SDRPlusPlus/pull/1115).

DuMOHsmol, 2023-07-02:
@AlexandreRouma, how do you think about allowing arbitrary `uiScale` fractions in config file without exposing new options in UI until everything is fixed?

For me it's currently the choice between something being misaligned and the whole app being at inconvenient scale, and I would rather pick the first option. Fun fact: I have actually used a private fork of SDR++ with all fonts scaled up before there was any scaling support. Everything was misaligned, but it still was better than squinting at everything.

bczhc, 2024-03-04:
I have a 23.8'', 2560x1440 display, and 150% will be just fine. I had forked sdrpp and modified it, adding my 150% option:
```diff
diff --git a/core/src/gui/menus/display.cpp b/core/src/gui/menus/display.cpp
index 59649ed..a9f2422 100644
--- a/core/src/gui/menus/display.cpp
+++ b/core/src/gui/menus/display.cpp
@@ -122,6 +122,7 @@ namespace displaymenu {
         // Define and load UI scales
         uiScales.define(1.0f, "100%", 1.0f);
         uiScales.define(2.0f, "200%", 2.0f);
+        uiScales.define(1.5f, "150%", 1.5f);
         uiScales.define(3.0f, "300%", 3.0f);
         uiScales.define(4.0f, "400%", 4.0f);
         uiScaleId = uiScales.valueId(style::uiScale);
```

and, used it for a long time, everything is fine. No font and widget issues etc, at least as yet.

Also btw, I just updated SDR++, and it crashes on launch. After a debug, the cause is: `uiScaleId = uiScales.valueId(style::uiScale);`. SDR++ cannot find a menu option corresponding to the 1.5 scaling factor defined in the config file. I think at least, we should allow users to use custom scaling factors by modifying the config file.

JVital2013, 2024-12-06:
+1 for this request. On my laptop 100% is too small, and 200% is too big. Call me Goldilocks but something like 125% or even 150% would be just right 😄.

A related feature that would be nice is the ability to auto-detect scaling, based on OS settings. The option in the UI is still nice as an override, but it's often better when an app scales itself without needing user intervention. Although - I am fully aware of the pain it is to make this work cross-platform with ImGui, like I did in SatDump a while back. It's certainly a lower priority request.

RAD750, 2024-12-06:
I'd also like a fractional scaling option. I've switched recently from the venerable Thinkpad X260 to a Fujitsu Lifebook that has a high resolution screen, 100% is too small and 200% way too big. The workaround is to use the hidpi daemon on Gnome to scale it to 125%, however this results in the text and other UI elements being fuzzy and soft. 
SatDump works rather well, in comparison.