AlexandreRouma/SDRPlusPlus #1118: How to install SDR++server

marrem, 2023-07-02:
I installed SDR++ on my Mac. I couldn't connect to my SoapyServer running on my Raspberry Pi connected to my SDRPlay RSPdx. This is because the Soapy source module is not included in the nightly build I installed. 

You suggested to other people raising this issue, to run SDR++server on the raspberry pi. I assume SDR++server connects to the sdrplay api service running on the Pi. 

But I can't find the SDR++server project anywhere. And the readme didn't provide any clue. If it is part of the sdr++ distribution, how do I install it on the Pi without having to install all kinds of dependencies needed for SDR++ GUI that I won't run there.

AlexandreRouma, 2023-07-02:
SDR++ server is part of SDR++. See page 58 of the manual https://www.sdrpp.org/manual.pdf

closed 2023-07-02 by AlexandreRouma

marrem, 2023-07-02:
Yep, I have found it. I did install all dependencies. It runs now an I can connect to it from my Mac. Thanks