AlexandreRouma/SDRPlusPlus #1119: Too large font on Android

andimik, 2023-07-05:
**Hardware**

Sony G8441 Android mobile phone 

```
Pixeldichte 	319 ppi
Auflösung 	720 x 1.280 Pixel
Diagonale Zoll 	4,6 Zoll
Diagonale mm 	116,8 mm 
```

- CPU: 
- RAM: 4 GB
- GPU: 
- SDR: (Remote or local? If remote, what protocol?)

**Software**
- Operating System Android 9

- SDR++: Version + Build date https://github.com/AlexandreRouma/SDRPlusPlus/releases/download/nightly/sdrpp.apk and https://github.com/AlexandreRouma/SDRPlusPlus/commit/9ae6a7440844a4c6f0f27c11dd4ffd0bde4f6be1

**Bug Description**
On my Android phone I can't use the app as the font is too large, see screenshot. 

**Steps To Reproduce**
1. Installation of app
2. Run it

**Screenshots**
[image]

AlexandreRouma, 2023-07-05:
the high-dpi scaling setting is literally visible in your screenshot, expand the width thof the menu change it to something smaller. The default scaling is meant for modern smartphone with much higher resolutions than this one.

closed 2023-07-05 by AlexandreRouma

andimik, 2023-07-05:
Ok, I could successfully increase the column width and set it to another, lower value.

andimik, 2023-07-05:
I did not know this was adjustable... 😁