AlexandreRouma/SDRPlusPlus #1123: Squelch introduces noise on AM

themeadery, 2023-07-05:
**Hardware**
- CPU: AMD Ryzen 7 5800X
- RAM: 64GB DDR4
- GPU: RTX 3050
- SDR: local RTL-SDR blog V3

**Software**
- Operating System: Windows 11 Pro 22H2 22631.1972
- SDR++: v1.1.0 (Built at 23:35:14, Jun 28 2023)

**Bug Description**
Squelch setting is introducing noise. When the squelch is set too low and it is not muting there isn't very much noise, but I still see some on the tuned frequency. When the squelch is set to just above the threshold to mute there is a weird rhythmic spike of noise about once every 500ms. When the squelch is set even higher there is some severe noise introduced around DC.

This is only happening when trying to listen to low frequency AM channels (ie my local airport). Interestingly, the guide to do this on RTL-SDR blog's website doesn't work for me. I get nothing on Direct Sampling Q branch mode, I have to turn off Direct Sampling to get these AM tx's. The AM tx's come in fine, but the noise introduced by the squelch overlayed onto them hurts intelligibility. 

**Steps To Reproduce**
1. Radio -> AM
2. tune to AM freq in the ~118 MHz range
3. Turn on squelch and adjust slider from too low and no muting to just above threshold and then higher, watch noise pattern on waterfall and hear it in speakers

**Screenshots**
[image]
Rhythmic noise when squelch at threshold

[image]
Squelch higher than necessary, lots of DC noise

[image]
Squelch lowered way down, not much noise, but there is still some. I get this in some other bands even with IQ Correction turned on. Not sure what it means, but it seems somewhat normal.

**Additional info**
Happens at various sampling rates: 2.048 MHz to 2.88 MHz

AlexandreRouma, 2023-07-05:
From your issue description I assume you're new to SDR and I encourage you to first read up on the technology to understand what you are doing. I'm not here to teach people SDR/radio basics but I guess here we go...

> the guide to do this on RTL-SDR blog's website doesn't work for me. I get nothing on Direct Sampling Q branch mode, I have to turn off Direct Sampling to get these AM tx's. The AM tx's come in fine, but the noise introduced by the squelch overlayed onto them hurts intelligibility.

AM is just modulation, Direct sampling has to do with which frequencies you're tuning to, not the modulation. You need direct sampling for frequencies under 25MHz, not for airband.

> tune to AM freq in the ~118 MHz range

Here's your problem, you're not tuned into an AM airband signal. And since you're zoomed in all the way I can't tell what you're tuned into precisely. You're either on the DC spike (again, read up on the rtl-sdr's quirks) or some random interference from your sound card which would explain why it has feedback from the output.

Having visible change on the FFT from any setting in the radio menu is **physically impossible** to happen in software. Data has long been through the FFT before it gets anywhere close to the quelch...

closed 2023-07-05 by AlexandreRouma

themeadery, 2023-07-05:
Yes, I am new to SDR. However, I have been listening to communications all the way from the 160 meter band to the 70 cm band, from all over the world to my local police. I've got your software pretty dialed at this point, it's great. This is the first time I saw this happen and it didn't do it on the other airport frequency, just this one. I noticed it was tied to the squelch settings.

I can listen to the planes talk to the tower just fine, but on this channel it's mixed with noise from something and changing the squelch setting in SDR++ has a direct and large impact on what that noise is doing. At the threshold it is a very specifically timed spike of noise. It doesn't do that on ANY other channel, just this one.

> You need direct sampling for frequencies under 25MHz, not for airband.

Alright, the guide I followed was wrong then. I knew something was up when direct sampling let me listen on 160m/80m, but not my airport band. Obviously it works in regular mode now.

> Here's your problem, you're not tuned into an AM airband signal. And since you're zoomed in all the way I can't tell what you're tuned into precisely. 

Lol, I don't want to teach the guy who wrote the software how to use it, but if you look at the top left it shows the frequency of 118.250 MHz (fine tuned for some offset from the plane I was tracking). And in the Radio section bandwidth is set to "7000". Maybe my screenshots aren't showing up or you forgot you can click on them? They are in 1440p, should be readable.

> You're either on the DC spike (again, read up on the rtl-sdr's quirks) or some random interference from your sound card which would explain why it has feedback from the output.

Well, I was able to tune that DC spike out using IQ Correction on every other band I've tooled around on....except this one it seems. And the squelch setting makes it go from manageable to out of control red hot. You are saying the software can't cause that, but I can sit here and drag the squelch bar around and watch it have a direct effect on the DC spike noise you are referring to. It gets so big it overwhelms the original signal I am listening to.

This could be from my sound card, but only on that one exact frequency? I guess I'll just give up. I bought this dongle specifically to listen to my local airport.

themeadery, 2023-07-05:
For posterity.

My only other airport frequency. Zero DC noise regardless of squelch setting. Crystal clear communications being received between tower and ground. Meh...

[image]

AlexandreRouma, 2023-07-05:
> Lol, I don't want to teach the guy who wrote the software how to use it, but if you look at the top left it shows the frequency of 118.250 MHz (fine tuned for some offset from the plane I was [...]

I don't think you understood what I meant. What I mean is if you're zoomed in on a signal, there is no context to it, I can't tell if it's right in the middle of the SDR's passband in which case it would for sure be the DC spike, if it's not right in the middle then it's some other signal.

You need to understand that just because you see a spike on the waterfall, it doesn't mean there's actually a real broadcast signal there. It can be interference from inside your room, or even inside the hardware itself, which for cheap dongles like that isn't uncommon.

> but I can sit here and drag the squelch bar around and watch it have a direct effect on the DC spike noise you are referring to. It gets so big it overwhelms the original signal I am listening to.

If it's some interference from your sound card, then obviously when you change the sound you're gonna see something change on the waterfall.

> This could be from my sound card, but only on that one exact frequency?

Yes. Try muting the audio and see if changing the squelch setting still have any effect whatsoever (and actually check for correlation...)

> I guess I'll just give up. I bought this dongle specifically to listen to my local airport.

Don't just look for random frequencies, you probably won't find it. Look up your local airport's frequency online.

themeadery, 2023-07-05:
> Don't just look for random frequencies, you probably won't find it. Look up your local airport's frequency online.

I'm not doing that. I've flown planes, I know where to find the official frequencies to tune in to, lol. I already heard them communicating on our only two channels. This is one of them.

To answer the bulk of you other comments:

[image]

Zoomed out all the way.

> Yes. Try muting the audio and see if changing the squelch setting still have any effect whatsoever (and actually check for correlation...)

I just figured it out though. It is my audio interface. If I am playing audio in another app all the interference goes away. It's only if nothing else is playing I get the interference and it has this curious behavior right on the squelch threshold. I'll save you the 3 screen shots that explain it (or a screen recording) and just say thank you anyway, but it was my PreSonus Studio 68c doing weird shit only when no other audio is playing. And yes, muting in your software makes the interference go away FWIW. So you were right, lol!

themeadery, 2023-07-06:
If anyone is searching google and wonders how I solved this. I decided to generate an inaudible audio tone with a signal generator. I used Audacity to test this idea out. I played with many settings and 100 Hz sine wave at 0.00001 Amplitude seemed to work the best. My first try was 22 kHz because I am old and can't hear that regardless of the amplitude, but it sprays out a different undesirable noise pattern into my antenna/SDR stick. Pink noise also seems to work pretty good, but I'd rather have a single tone I think. Although, pink noise reminds me of the old BIOS options called "spread spectrum" to reduce EMI. I need to keep playing with this to make sure I get the lowest noise signature and don't damage my nice studio monitors. Shouldn't damage them, since the amplitude is like 1 dBFS, lol.

[image]

Above: You can see where I start playing the tone, the huge DC spike in the center of the tuned frequency (that's hard to see zoomed out) and all the harmonics up and down virtually disappear and turn into other lower amplitude noise. I think this will clean up my whole spectrum for other RX-ing too!!

[image]

Above: Audacity settings that worked best for me. Note: One more zero on the amplitude didn't quite work. Was just barely at the lower edge of the DAC resolution or something. Running 48 kHz 24-bit mode on my PreSonus software and Audacity is at 48 kHz 32-bit float.

My best guess is there is some sort of "squelch" on my audio interface that tries to make the speakers ultra quiet when no signal is being received on the output stage. This activates some sort of muting circuit (transistor switching on/off?) that then sprays interference out of one of the connectors or USB. There could be dozens of other possible sources, but playing an audio frequency of any sort kills this noise. You can see this noise in awesome harmonic patterns up and down the RF spectrum when you "turn it on and off" by playing audio. Could also just be a low load on the SMPS for the interface make it unhappy, so when you load it a bit it cleans up.

I just so happened to stumble on this because of my local airport's tower frequency is right at one of the harmonics of that noise my audio interface makes. Like smack dab bang on the center of it. What luck! The next dumb luck move was noticing that when I was playing Spotify music all the interference went away. So if you don't want to play music AND listen to your SDR software, then you need to generate a really quiet tone and try to push it into one side of the audio spectrum.

I suppose this feature could be built right in to SDR++ with a plugin or something, but I'm busy with too many other projects to give it a shot. Cheers everyone.