AlexandreRouma/SDRPlusPlus #1124: Installer for Windows

AsciiWolf, 2023-07-06:
Consider adding an installer, ideally in MSI format (or EXE with silent install support), for the Windows build of SDR++.

I am planning to work on a [WinGet](https://en.wikipedia.org/wiki/Windows_Package_Manager) script (see [this ticket](https://github.com/microsoft/winget-pkgs/issues/111415)) for SDR++ and MSI installer would make this way easier. :-)

AlexandreRouma, 2023-07-06:
I see no reason to package SDR++ this way. It is meant to be portable, installing it would not only complicate things but is much less flexible compared to a fully portable version like the way it's currently distributed. Literally just unzip where you want it install, doesn't get any simpler.

closed 2023-07-06 by AlexandreRouma

AsciiWolf, 2023-07-06:
Not everyone wants to use portable applications. Also, automated deployments (and updates) of portable apps are a nightmare. Speaking of automated deployments, I found out that WinGet does not support self contained zips with included DLLs, so no MSI/EXE means that it probably won't be possible to create a WinGet install script for SDR++. :-(

Please, reconsider this. You already ship *.deb packages for Debian/Ubuntu and have install target for Unix/Linux in the CMakeLists file. If you do not want to spend time on this, that's perfectly fine, but please at least consider keeping this ticket open so other people can find it and possibly start working on this. Thanks!

AlexandreRouma, 2023-07-07:
> Not everyone wants to use portable applications. 

Not everyone is right.

> automated deployments (and updates) of portable apps are a nightmare.

Literally just extract the zip.

> I found out that WinGet does not support self contained zips with included DLLs

Maybe microsoft should add this basic feature of every package manager ever then?

> won't be possible to create a WinGet install script for SDR++. :-(

Not a big loss considering nobody uses it anyway and I don't plan to support it. Only official/supported windows builds are on this github page.

> You already ship *.deb packages for Debian/Ubuntu and have install target for Unix/Linux in the CMakeLists file.

Those operating systems require packaging to distribute software correctly. Windows explicitly does not and wasting space, install time and maintainer time by making an installer for a few dozen files that can be anywhere is just silly. I won't reconsider and won't support any third party installer.