AlexandreRouma/SDRPlusPlus #1126: No import or export dialog for Android

rfbio, 2023-07-07:
**Hardware**
- CPU: Octa-core (4x2.3 GHz Cortex-A73 & 4x1.7 GHz Cortex-A53)
- RAM: 4GB 
- GPU: Mali-G72 MP3
- SDR: local

**Software**
- Operating System: Android 13
- SDR++: v1.1.0, Built on: Jul 2, 2023

**Bug Description**
No import or export dialogue for Frequency Manager

**Steps To Reproduce**
1. Attach SDR 
2. Open SDR++
3. Refresh source & "Play" (or Refresh & leave idle)
4. Open Freq Man tab
5. Touch/click import OR export
6. No dialog appears to follow through

**Screenshots**
None

**Additional info**
Device: Android Samsung Galay Tab S6 Lite

AlexandreRouma, 2023-07-07:
Not a bug, it's just not supported yet on android.

rfbio, 2023-07-07:
> 

Good to know. Thank you for your time!

closed 2023-07-07 by rfbio