AlexandreRouma/SDRPlusPlus #1133: MacOS - Audio sinks missing with newest nightly build

michaelgeorge001, 2023-07-11:
**Hardware**
Mac mini M1
16GB RAM
Nooelec NESDR SMArTee v2 SDR

**Software**
MacOS 12.6.7

**Bug Description**
Audio sinks missing with newest nightly build

Audio sinks display and operate properly with the Jul 9, 2023 nightly build and earlier.
Audio sinks are missing on the Jul 10, 2023 nightly build.

**Steps To Reproduce**
Using the same config file, and toggling back and forth between both nightly builds produces the following results.

Jul 9 version - Audio sinks display and audio is heard.
July 10 version - Audio sinks do not display, and obviously no audio is heard.

**Screenshots**

Here are the relevant screen shots:
[image]

[image]

Unless I'm missing a new config setting, this doesn't seem to be correct behaviour.

-Mike-

michaelgeorge001, 2023-07-11:
In addition the new Streams module is also inoperable in the newest version as well.
[image]

-Mike-

AlexandreRouma, 2023-07-11:
you downloaded some random CI build from the `new_sinks` branch, **NOT** a nightly.

Nightly builds are available at https://www.sdrpp.org/nightly

closed 2023-07-11 by AlexandreRouma

michaelgeorge001, 2023-07-11:
Thank you, my mistake.

However, it wasn't just some random CI build, it was the newest.

Regardless, it was my error in not confirming that a nightly build was created from it or that it was from the master branch.

I will pull from the nightly build link from this point forward, or compile it myself.

Thanks again.

-Mike-