AlexandreRouma/SDRPlusPlus #1139: sdr-rtl lost

biscegliem, 2023-07-15:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: 
- RAM:
- GPU: 
- SDR: (Remote or local? If remote, what protocol?)

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

**Bug Description**
A clear description of the bug.

**Steps To Reproduce**
1. ...
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

biscegliem, 2023-07-15:
in Ubuntu Jammy , last release, I downloaded sdr++ nightly last release , the rtl-sdr source is not available, only RTL-tcp have you the same problem?

sizzlik, 2023-07-17:
you probably closed it. Just go to the Module Manager and start it again?

biscegliem, 2023-07-18:
Hello sizzlik, unfortunately the problem is that in the Module Manager the rtl sdr source is lost ...

sizzlik, 2023-07-18:
[rtl_sdr_source.zip](https://github.com/AlexandreRouma/SDRPlusPlus/files/12079674/rtl_sdr_source.zip)

Here is the module dll. Copy into the modules folder. Or download the project again, unzip, inside the modules folder is the dll aswell.

sizzlik, 2023-07-18:
But the module file is there? (for example /usr/lib/sdrpp/plugins/rtl_sdr_source.so on linux)

sizzlik, 2023-07-18:
Good luck. If you got it fixed, please come back and close the issue as it seems nobody is moderating here.

biscegliem, 2023-07-19:
Hello sizzlik, thanks for your support, finally the problem was solved. The issue was from the ubuntu software installer, I installed synaptic and the sdrpp installation was finally done without problems.

closed 2023-07-20 by biscegliem