AlexandreRouma/SDRPlusPlus #1142: SDR receiver detection on Android

AWatGit, 2023-07-19:
The Airspy HF+ Discovery is sometimes recognised after pressing Refresh and sometimes not. Are there any improvements to be expected?

AlexandreRouma, 2023-09-13:
> Are there any improvements to be expected?

No, I've decided I like my software being buggy (sarcasm)...

paccerdk, 2024-08-12:
Having the same problem with my AirSpy R1 on Android, eventually got it to work after restarting SDR++ a bunch of times - let me know if there is anything i can do to help fix it

Flo250578, 2024-09-11:
> The Airspy HF+ Discovery is sometimes recognised after pressing Refresh and sometimes not. Are there any improvements to be expected?

It is possible that you have this problem from  the Android device.  The same happens with mine RedmiNote 9 and SDR-RTL v3.  Once it works and 3 times it doesn't.  I noticed this especially after upgrading to Android 12 and MIUI 13. At first I thought it was a problem with SDR++, but I installed SDR Touch which comes with a dedicated driver and does exactly the same thing.  I think there is an integrated service in the stock ROM that blocks the triggering of the OTG on devices like sdr.  With SD reader and wifi mouse it works ok.  It should be noted that I tested SDR++ on the Galaxy Note9 and Galaxy A30 and it works flawlessly all the time.  I will try to downgrade Redmi Note9 to android 11 then to 10. I hope it will work because I bought this device intentionally for a standalone panadapter.

Mraanderson, 2024-11-23:
Using a Google Pixel 3a XL (LineageOS 21 Android 14) and RTL-SDR I'm finding the prompt to allow use of the driver flashes up half way through SDR++ loading. Swip out of the app, force stop and open fixes the connection. I should give that refresh button a go. Could do with a slow down on launch or a pause button.