AlexandreRouma/SDRPlusPlus #1143: SDR++ crashes on launching

Pbaodoge, 2023-07-19:
**Hardware**
- CPU: Broadcom BCM2711
- RAM: 4 GB
- GPU: RPI 4 onboard GPU
- SDR: local

**Software**
- Operating System: Raspberry Pi OS - Debian GNU/Linux 11 (bullseye)
- SDR++: nightly, built date: 19/7

**Bug Description**
### I tried to build the nightly version of SDR++, it builds successfully of course and everything went smoothly until I tried to launch the software, and I got this error:
```sh
pi@raspberrypi:~/Desktop/sdr/build $ sdrpp
[2023-07-19 23:21:16.082] [info] SDR++ v1.1.0
[2023-07-19 23:21:16.083] [info] Loading config
[2023-07-19 23:21:16.084] [warning] ConfigManager locked, waiting...
[2023-07-19 23:21:16.701] [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
[2023-07-19 23:21:16.701] [info] OpenGL 3.0 was not supported
[2023-07-19 23:21:16.812] [info] Using OpenGL 3.1 ES
[2023-07-19 23:21:17.116] [info] Loading icons
[2023-07-19 23:21:17.150] [info] Loading band plans
[2023-07-19 23:21:17.163] [info] Loading band plans color table
[2023-07-19 23:21:17.165] [error] Menu element is missing name key
[2023-07-19 23:21:17.183] [info] Loading modules
[2023-07-19 23:21:17.184] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2023-07-19 23:21:17.189] [info] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[2023-07-19 23:21:17.192] [info] Loading /usr/lib/sdrpp/plugins/m17_decoder.so
[2023-07-19 23:21:17.194] [error] Couldn't load /usr/lib/sdrpp/plugins/m17_decoder.so.
[2023-07-19 23:21:17.194] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so
[2023-07-19 23:21:17.230] [info] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[2023-07-19 23:21:17.234] [info] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[2023-07-19 23:21:17.238] [info] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so
[2023-07-19 23:21:17.249] [error] Couldn't load /usr/lib/sdrpp/plugins/new_portaudio_sink.so.
[2023-07-19 23:21:17.249] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[2023-07-19 23:21:17.253] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2023-07-19 23:21:17.258] [info] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[2023-07-19 23:21:17.260] [error] Couldn't load /usr/lib/sdrpp/plugins/limesdr_source.so.
[2023-07-19 23:21:17.260] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[2023-07-19 23:21:17.277] [info] Loading /usr/lib/sdrpp/plugins/weather_sat_decoder.so
[2023-07-19 23:21:17.297] [error] Couldn't load /usr/lib/sdrpp/plugins/weather_sat_decoder.so.
[2023-07-19 23:21:17.297] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2023-07-19 23:21:17.321] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[2023-07-19 23:21:17.326] [info] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[2023-07-19 23:21:17.328] [error] Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so.
[2023-07-19 23:21:17.328] [info] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[2023-07-19 23:21:17.333] [info] Loading /usr/lib/sdrpp/plugins/bladerf_source.so
[2023-07-19 23:21:17.335] [error] Couldn't load /usr/lib/sdrpp/plugins/bladerf_source.so.
[2023-07-19 23:21:17.335] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2023-07-19 23:21:17.342] [info] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[2023-07-19 23:21:17.358] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2023-07-19 23:21:17.379] [info] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[2023-07-19 23:21:17.397] [info] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[2023-07-19 23:21:17.403] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2023-07-19 23:21:17.407] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2023-07-19 23:21:17.429] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2023-07-19 23:21:17.433] [info] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[2023-07-19 23:21:17.457] [warning] ConfigManager locked, waiting...
[2023-07-19 23:21:17.457] [info] Loading /usr/lib/sdrpp/plugins/scanner.so
[2023-07-19 23:21:17.461] [info] Initializing Airspy Source (airspy_source)
[2023-07-19 23:21:17.494] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2023-07-19 23:21:17.516] [info] Initializing Audio Sink (audio_sink)
[2023-07-19 23:21:17.526] [info] Initializing Audio Source (audio_source)
[2023-07-19 23:21:17.529] [error] Module 'audio_source' doesn't exist
[2023-07-19 23:21:17.529] [info] Initializing BladeRF Source (bladerf_source)
[2023-07-19 23:21:17.533] [error] Module 'bladerf_source' doesn't exist
[2023-07-19 23:21:17.534] [info] Initializing File Source (file_source)
[2023-07-19 23:21:17.547] [info] Initializing Frequency Manager (frequency_manager)
[2023-07-19 23:21:17.561] [info] Initializing HackRF Source (hackrf_source)
[2023-07-19 23:21:17.577] [info] Initializing Hermes Source (hermes_source)
[... 59 lines cut ...]

In file included from /home/pi/Desktop/sdr/misc_modules/frequency_manager/src/main.cpp:12:
/home/pi/Desktop/sdr/core/src/gui/file_dialogs.h:1654:8: note: declared here
 1654 | inline open_file::open_file(std::string const &title,
      |        ^~~~~~~~~
/home/pi/Desktop/sdr/misc_modules/frequency_manager/src/main.cpp:547:137: warning: ‘pfd::save_file::save_file(const string&, const string&, const std::vector<std::__cxx11::basic_string<char> >&, bool)’ is deprecated: Use pfd::opt::force_overwrite instead of confirm_overwrite [-Wdeprecated-declarations]
  547 | pfd::save_file("Export bookmarks", "", { "JSON Files (*.json)", "*.json", "All Files", "*" }, true);
      |                                                                                                   ^

In file included from /home/pi/Desktop/sdr/misc_modules/frequency_manager/src/main.cpp:12:
/home/pi/Desktop/sdr/core/src/gui/file_dialogs.h:1678:8: note: declared here
 1678 | inline save_file::save_file(std::string const &title,
      |        ^~~~~~~~~
/home/pi/Desktop/sdr/misc_modules/frequency_manager/src/main.cpp: In member function ‘void FrequencyManagerModule::exportBookmarks(std::string)’:
/home/pi/Desktop/sdr/misc_modules/frequency_manager/src/main.cpp:790:30: warning: ‘std::ostream& nlohmann::operator>>(const nlohmann::basic_json<>&, std::ostream&)’ is deprecated: Since 3.0.0; use operator<<(std::ostream&, const basic_json&) [-Wdeprecated-declarations]
  790 |         exportedBookmarks >> fs;
      |                              ^~
In file included from /home/pi/Desktop/sdr/core/src/module.h:4,
                 from /home/pi/Desktop/sdr/misc_modules/frequency_manager/src/main.cpp:3:
/home/pi/Desktop/sdr/core/src/json.hpp:22968:26: note: declared here
22968 |     friend std::ostream& operator>>(const basic_json& j, std::ostream& o)
      |                 
```
I greatly appreciate any assistance with this issue. 
Thanks!

AlexandreRouma, 2023-07-19:
you either updated from a very old version or had a crash that corrupted a config file. Reset the RTL-TCP source config by running `rm ~/.config/sdrpp/rtl_tcp_config.json`.

Pbaodoge, 2023-07-20:
It works without runtime errors this time, thank u so much!

closed 2023-07-20 by Pbaodoge