AlexandreRouma/SDRPlusPlus #1144: Export ALL Frequency manager Lists at once

gprime31, 2023-07-21:
**Feature description**
The ability to export the whole freq. list(s) at once.

Or maybe I missed somthing, but having to shift click one at a time to select the whole list and export is a pain.

Daniel-from-Germany, 2023-07-21:
Dont need it...
look for the file in the root directory: frequency_manager_config.json
then use a converter: https://data.page/json/csv

gprime31, 2023-07-21:
> Dont need it... look for the file in the root directory: frequency_manager_config.json then use a converter: https://data.page/json/csv

I don't want to do all that, the export works fine the way it is, maybe just enable Ctrl+a to select all and I can export all of them easily. The idea is to export all at once to have a backup-freq.json that I can put in a shared folder then, import to all my other devices running sdrpp.

Daniel-from-Germany, 2023-07-21:
I do this:

```
@echo off
set "sourceDir=C:\PATH-TO\sdrpp_windows_x64"
set "backupDir=C:\PATH-TO\sdrpp_windows_x64\backup-json-%DATE%-%TIME:~0,2%-%TIME:~3,2%-%TIME:~6,2%"
set "programm=C:\PATH-TO\sdrpp_windows_x64\sdrpp.exe"

rem Create a Backupfolder
if not exist "%backupDir%" (
    mkdir "%backupDir%"
)

rem Copy all json Files to Backupfolder
for %%F in ("%sourceDir%\*.json") do (
    copy "%%F" "%backupDir%\"
)

rem Copy my own default frequency file bevor start the sdr
xcopy /y "%sourceDir%\frequency_manager_config-my.json" "%sourceDir%\frequency_manager_config.json"

rem Start sdrpp.exe
start "" "%programm%"

```

Its a easy way to handle all what you want, and its automatically 
have fun ;)

gprime31, 2023-07-21:
Same for Android and Windows 11?